zephyrproject-rtos/zephyr #58930: [QEMU] mqtt azure sample crash "Access violation: supervisor thread not allowed to write"

n3rd4i, 2023-06-07:
**Describe the bug**
QEMU build of `samples/net/cloud/mqtt_azure` application should not crash

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b qemu_x86 samples/net/cloud/mqtt_azure -p -- -DCONF_FILE="prj.conf overlay-qemu_x86.conf"
2. west build -t run
3. See error

**Expected behavior**
QEMU application should not crash.

**Impact**
showstopper

**Logs and console output**
```
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..

*** Booting Zephyr OS build zephyr-v3.3.0-2068-g86da50c9861a ***
[00:00:00.010,000] <wrn> net_sock_tls: No entropy device on the system, TLS communication is insecure!
[00:00:00.020,000] <inf> net_config: Initializing network
[00:00:00.020,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.020,000] <dbg> mqtt_azure: main: Waiting for network to setup...
[00:00:00.060,000] <inf> mqtt_azure: DNS resolved for REDACTED.azure-devices.net:8883
[00:00:00.060,000] <dbg> mqtt_azure: try_to_connect: attempting to connect...
[00:00:00.260,000] <inf> net_mqtt: Connect completed
[00:00:02.720,000] <dbg> mqtt_azure: mqtt_event_handler: MQTT client connected!
[00:00:02.730,000] <inf> mqtt_azure: SUBACK packet id: 1
uart:~$
[00:00:16.740,000] <dbg> mqtt_azure: publish_timeout: mqtt_publish OK
[00:00:17.180,000] <err> os: Page fault at address (nil) (error code 0x2)
[00:00:17.190,000] <err> os: Linear address not present in page tables
[00:00:17.190,000] <err> os: Access violation: supervisor thread not allowed to write
[00:00:17.190,000] <err> os: PTE: not present
[00:00:17.190,000] <err> os: EAX: 0x00175720, EBX: 0x00000004, ECX: 0x00000000, EDX: 0x0011be0c
[00:00:17.190,000] <err> os: ESI: 0x00152884, EDI: 0x00000000, EBP: 0x00172088, ESP: 0x00172084
[00:00:17.190,000] <err> os: EFLAGS: 0x00000087 CS: 0x0008 CR3: 0x0017c000
[00:00:17.190,000] <err> os: call trace:
[00:00:17.190,000] <err> os: EIP: 0x00143982
[00:00:17.190,000] <err> os:      0x0014190c (0x152894)
[00:00:17.190,000] <err> os:      0x001128d0 (0x0)
[00:00:17.190,000] <err> os:      0x0011f232 (0x15280c)
[00:00:17.190,000] <err> os:      0x001252bf (0x152c88)
[00:00:17.190,000] <err> os:      0x0012569d (0x15280c)
[00:00:17.190,000] <err> os:      0x0012194b (0x11d8cd)
[00:00:17.190,000] <err> os:      0x001223ab (0x171aac)
[00:00:17.190,000] <err> os:      0x0011cfb0 (0x171aac)
[00:00:17.190,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:17.190,000] <err> os: Current thread: 0x152ba0 (rx_q[0])
[00:00:17.190,000] <err> os: Halting system
uart:~$ FAILED: zephyr/CMakeFiles/run_qemu /home/user/zephyrproject/zephyr/build/zephyr/CMakeFiles/run_qemu
cd /home/user/zephyrproject/zephyr/build && /home/user/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-i386 -m 4 -cpu qemu32,+nx,+pae -machine q35 -device isa-debug-exit,iobase=0xf4,iosize=0x04 -no-reboot -nographic -no-acpi -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -serial unix:/tmp/slip.sock -kernel /home/user/zephyrproject/zephyr/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/user/zephyrproject/zephyr/build --target run
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK,
 - Commit SHA or Version used: zephyr-v3.3.0-2068-g86da50c9861a

n3rd4i, 2023-06-12:
hi @rlubos it seems like that, this is the complete [updated] log:
```
[00:00:11.310,000] <dbg> mqtt_azure: publish_timeout: mqtt_publish OK
[00:00:11.310,000] <err> os: EAX: 0x00176898, EBX: 0x00000246, ECX: 0x00000000, EDX: 0x0016dd80
[00:00:11.310,000] <err> os: ESI: 0x00152040, EDI: 0x00152040, EBP: 0x00176c50, ESP: 0x00176c44
[00:00:11.310,000] <err> os: EFLAGS: 0x00000003 CS: 0x0008 CR3: 0x0017d000
[00:00:11.310,000] <err> os: call trace:
[00:00:11.310,000] <err> os: EIP: 0x00141910F
[00:00:11.310,000] <err> os:      0x001424cb (0x216)
[00:00:11.310,000] <err> os:      0x0010ec22 (0x16dd80)
[00:00:11.310,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:11.310,000] <err> os: Current thread: 0x16dd80 (sysworkq)
[00:00:11.320,000] <err> os: Halting system
uart:~$ FAILED: zephyr/CMakeFiles/run_qemu /home/user/zephyrproject/zephyr/build/zephyr/CMakeFiles/run_qemu 
cd /home/user/zephyrproject/zephyr/build && /home/user/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-i386 -m 4 -cpu qemu32,+nx,+pae -machine q35 -device isa-debug-exit,iobase=0xf4,iosize=0x04 -no-reboot -nographic -no-acpi -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -serial unix:/tmp/slip.sock -kernel /home/user/zephyrproject/zephyr/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/user/zephyrproject/zephyr/build --target run

```

rlubos, 2023-06-13:
Ok. so it's system workqueue. Can you try to increase `CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE` and see if it solves the problem? It should help with the stack overflow, but it would be good to confirm it was the only problem.

n3rd4i, 2023-06-13:
I've used `CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048` and this seems to resolve the issue! 

Does this mean that the ticket can be set to resolved if the example configuration has this stack size increased?

rlubos, 2023-06-13:
> Does this mean that the ticket can be set to resolved if the example configuration has this stack size increased?

I'd say so, if there are no other problems. Could you send a PR that updates the sample config file?

closed 2023-06-14 by nashif