zephyrproject-rtos/zephyr #58931: tests/drivers/dac loopback failed due to un-initialised dac_channel_cfg structure

FRASTM, 2023-06-07:
**Describe the bug**
Testing the tests/drivers/dac/dac_loopback/  fails on several stm32 boards
```
START - test_dac_loopback                                                       
                                                                                
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/dac/dac_loopback/src/t)
Value 2091 read from ADC does not match expected range.                         
 FAIL - test_dac_loopback in 0.031 seconds       
```                               

Please also mention any information which could help others to understand
the problem you're facing:
The PR https://github.com/zephyrproject-rtos/zephyr/pull/57730  is checking the `channel_cfg->buffered`
This value must be initialised.

**To Reproduce**
Steps to reproduce the behavior:
1.west build -p auto -b nucleo_wl55jc tests/drivers/dac/dac_loopback/
2. west flash
3. See error

**Expected behavior**
```
TESTSUITE dac succeeded
------ TESTSUITE SUMMARY START ------
SUITE PASS - 100.00% [dac]: pass = 1, fail = 0, skip = 0, total = 1 duration = s
 - PASS - [dac.test_task_write_value] duration = 0.001 seconds
```
------ TESTSUITE SUMMARY END ------

**Impact**
test and CI failure

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain Zephyr SDK 0.16.1
 - Commit SHA 7b2034aaecc4cb2a261973b10b2fa608b29d398c

martinjaeger, 2023-06-07:
@FRASTM Thanks for finding and fixing this issue.

Do you have an explanation why buffering is required? I would have guessed that the ADC input impedance is high enough such that buffering the DAC output is not necessary.

FRASTM, 2023-06-07:
> @FRASTM Thanks for finding and fixing this issue.
> 
> Do you have an explanation why buffering is required? I would have guessed that the ADC input impedance is high enough such that buffering the DAC output is not necessary.

I just see considered the outpin  "Connected to external pin and to on chip-peripherals (such as comparators)" 
[image]

martinjaeger, 2023-06-07:
Ok, so it looks like we should instead use `LL_DAC_SetOutputConnection` to make sure the DAC is connected to an external pin. Also with buffer disabled we should be able to get an output on the external pin.

The root cause of the failing test is not the buffer, but the output connection. We just didn't realize before when the output buffer was always enabled.

closed 2023-06-08 by nashif