zephyrproject-rtos/zephyr #58933: drivers: uart: async_api: drivers.uart.async_api.nrf_uart for nrf9160dk_nrf9160 fails

katgiadla, 2023-06-07:
**Describe the bug**
Tests `drivers.uart.async_api.nrf_uart` for `nrf9160dk_nrf9160` fails.

**To Reproduce**
Steps to reproduce the behavior:

1. have `nrf9160dk_nrf9160` connected
2. connect pins follow `tests/drivers/uart/uart_async_api/boards/nrf9160dk_nrf9160.overlay`
3. go to your zephyr dir
4. call `$ZEPHYR_BASE/scripts/twister -vv --inline-logs --fixture gpio_loopback -T ./tests/drivers/uart/uart_async_api/ --device-testing --hardware-map map.yml --platform nrf9160dk_nrf9160 -j 1`
5. See handler.log and console output

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.4.0-rc2-47-g42d7d114aeb3 ***
Running TESTSUITE uart_async_chain_read
===================================================================
START - test_chained_read
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:318: uart_async_chain_read_test_chained_read: (k_sem_take(&rx_disabled, K_MSEC(10)) equal to 0)
RX_DISABLED occurred
FAIL - test_chained_read in 0.022 seconds
===================================================================
TESTSUITE uart_async_chain_read failed.
Running TESTSUITE uart_async_chain_write
===================================================================
START - test_chained_write
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:719: uart_async_chain_write_test_chained_write: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_chained_write in 0.123 seconds
===================================================================
TESTSUITE uart_async_chain_write failed.
Running TESTSUITE uart_async_double_buf
===================================================================
START - test_double_buffer
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:393: uart_async_double_buf_test_double_buffer: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_double_buffer in 0.121 seconds
===================================================================
TESTSUITE uart_async_double_buf failed.
Running TESTSUITE uart_async_long_buf
===================================================================
START - test_long_buffers
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:792: uart_async_long_buf_test_long_buffers: (k_sem_take(&rx_rdy, K_MSEC(200)) not equal to 0)
RX_RDY timeout
FAIL - test_long_buffers in 0.264 seconds
===================================================================
TESTSUITE uart_async_long_buf failed.
Running TESTSUITE uart_async_multi_rx
===================================================================
START - test_multiple_rx_enable
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:187: uart_async_multi_rx_test_multiple_rx_enable: (k_sem_take(&rx_disabled, K_MSEC(100)) not equal to -EAGAIN)
RX_DISABLED not expected at this point
FAIL - test_multiple_rx_enable in 0.124 seconds
===================================================================
TESTSUITE uart_async_multi_rx failed.
Running TESTSUITE uart_async_read_abort
===================================================================
START - test_read_abort
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:485: uart_async_read_abort_test_read_abort: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_read_abort in 0.121 seconds
===================================================================
TESTSUITE uart_async_read_abort failed.
Running TESTSUITE uart_async_single_read
===================================================================
START - test_single_read
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:136: uart_async_single_read_test_single_read: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_single_read in 0.221 seconds
===================================================================
TESTSUITE uart_async_single_read failed.
Running TESTSUITE uart_async_timeout
===================================================================
START - test_forever_timeout
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:649: uart_async_timeout_test_forever_timeout: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_forever_timeout in 3.121 seconds
===================================================================
TESTSUITE uart_async_timeout failed.
Running TESTSUITE uart_async_write_abort
===================================================================
START - test_write_abort
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:569: uart_async_write_abort_test_write_abort: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_write_abort in 0.121 seconds
===================================================================
TESTSUITE uart_async_write_abort failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [uart_async_chain_read]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.022 seconds
- FAIL - [uart_async_chain_read.test_chained_read] duration = 0.022 seconds
SUITE FAIL -   0.00% [uart_async_chain_write]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.123 seconds
- FAIL - [uart_async_chain_write.test_chained_write] duration = 0.123 seconds
SUITE FAIL -   0.00% [uart_async_double_buf]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.121 seconds
- FAIL - [uart_async_double_buf.test_double_buffer] duration = 0.121 seconds
SUITE FAIL -   0.00% [uart_async_long_buf]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.264 seconds
- FAIL - [uart_async_long_buf.test_long_buffers] duration = 0.264 seconds
SUITE FAIL -   0.00% [uart_async_multi_rx]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.124 seconds
- FAIL - [uart_async_multi_rx.test_multiple_rx_enable] duration = 0.124 seconds
SUITE FAIL -   0.00% [uart_async_read_abort]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.121 seconds
- FAIL - [uart_async_read_abort.test_read_abort] duration = 0.121 seconds
SUITE FAIL -   0.00% [uart_async_single_read]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.221 seconds
- FAIL - [uart_async_single_read.test_single_read] duration = 0.221 seconds
SUITE FAIL -   0.00% [uart_async_timeout]: pass = 0, fail = 1, skip = 0, total = 1 duration = 3.121 seconds
- FAIL - [uart_async_timeout.test_forever_timeout] duration = 3.121 seconds
SUITE FAIL -   0.00% [uart_async_write_abort]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.121 seconds
- FAIL - [uart_async_write_abort.test_write_abort] duration = 0.121 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 7e5545a9ddeca40a839511798b6d4de9
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.5 LTS
- Toolchain Zephyr 0.16.0 
- Commit SHA or Version used: zephyr-v3.4.0-rc2-47-g42d7d1

**Additional context**
Pin set for `nrf9160dk_nrf9160`:
```
psels = <NRF_PSEL(UART_TX, 0, 10)>,
		<NRF_PSEL(UART_RX, 0, 11)>;
```

carlescufi, 2023-06-07:
This might be the same issue as #58934

katgiadla, 2023-06-07:
It was mistake with connecting jump wire. Currently, problem does not occure.

closed 2023-06-07 by katgiadla