zephyrproject-rtos/zephyr #58934: drivers: regulator: fixed: drivers.regulator.fixed for nrf9160dk_nrf9160 fails

katgiadla, 2023-06-07:
**Describe the bug**
Tests `drivers.regulator.fixed` for `nrf9160dk_nrf9160` fails.

**To Reproduce**
Steps to reproduce the behavior:

1. have `nrf9160dk_nrf9160` connected
2. connect pins follow `tests/drivers/regulator/fixed/boards/nrf9160dk_nrf9160.overlay`
3. go to your zephyr dir
4. call `$ZEPHYR_BASE/scripts/twister -vv --inline-logs --fixture gpio_loopback -T ./tests/drivers/regulator/fixed --device-testing --hardware-map map.yml --platform nrf9160dk_nrf9160 -j 1`
5. See handler.log and console output

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.4.0-rc2-47-g42d7d114aeb3 ***
Running TESTSUITE regulator_fixed
===================================================================
START - test_enable_disable
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/regulator/fixed/src/main.c:28: regulator_fixed_test_enable_disable: ret not equal to 1
FAIL - test_enable_disable in 0.015 seconds
===================================================================
TESTSUITE regulator_fixed failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [regulator_fixed]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.015 seconds
- FAIL - [regulator_fixed.test_enable_disable] duration = 0.015 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: dee8ab7c4f16a452f501f82af54bc0e0
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**

- OS: Ubuntu 20.04.5 LTS
- Toolchain Zephyr 0.16.0
- Commit SHA or Version used: zephyr-v3.4.0-rc2-47-g42d7d1

**Additional context**
Pin set for `nrf9160dk_nrf9160`:
```
enable-gpios = <&gpio0 10 GPIO_ACTIVE_HIGH>;
check-gpios = <&gpio0 11 (GPIO_ACTIVE_HIGH | GPIO_PULL_DOWN)>;
```

gmarull, 2023-06-07:
This sounds like aan _impedance_ mismatch between setup wiring and what's defined in the test overlay for that board.

carlescufi, 2023-06-07:
> This sounds like an _impedance mismatch_ between setup wiring and what's defined in the test overlay for that board.

Can you test it out to see if that is the case @katgiadla?

katgiadla, 2023-06-07:
It was mistake with connecting jump wire. Currently, problem does not occure.

closed 2023-06-07 by katgiadla