zephyrproject-rtos/zephyr #58936: Userspace: Disabled APIs have impact on binary size

erwango, 2023-06-07:
**Describe the bug**
Seen when investigating https://github.com/zephyrproject-rtos/zephyr/issues/58416

When building an application with `CONFIG_USERSPACE=y`, non used APIs have impact on final binary size.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b nucleo_l073rz samples/hello_world -- -DCONFIG_USERSPACE=y
```
[181/181] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       29592 B       192 KB     15.05%
             RAM:        7544 B        20 KB     36.84%
        IDT_LIST:          0 GB         2 KB      0.00%
```
2. Make some modification in an API disabled. Eg, in `include/zephyr/drivers/counter.h`
```
- __syscall int counter_start(const struct device *dev);
```
Note: Line should be removed, commenting or `#if 0` have no effect
3. Build again: `west build -p always -b nucleo_l073rz samples/hello_world -- -DCONFIG_USERSPACE=y`
5. Compare Flash footprint. It decreased by 4 bytes.
```
[181/181] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       29588 B       192 KB     15.05%
             RAM:        7544 B        20 KB     36.84%
        IDT_LIST:          0 GB         2 KB      0.00%
```

**Expected behavior**
Disabled APIs should not have impact on final binary size.

**Impact**
In USERSPACE config binaries are bloated with disabled/non supported APIs.
In long term, some small devices might not support USERSPACE anymore because too much APIs are available in Zephyr.

erwango, 2023-06-07:
Reducing bug priority as potential gain is not huge.
```
$ grep -r __syscall include/zephyr/drivers/  | wc -l
299
```

nashif, 2023-06-07:
see #58351

erwango, 2023-06-07:
> see #58351

The power of community work!

Will test on https://github.com/zephyrproject-rtos/zephyr/issues/58416

erwango, 2023-06-07:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/58351

PR #58351 (2023-06-17): kernel: syscalls: no need to include all syscalls in binary
The syscall generation phase parses all header files to look for potential syscalls, and emits all the relevant files to enable syscalls. However, this results in all the syscall marshalling functions being included in the final binary. This is due to these functions being referred to inside the dispatch list, resulting in ineffective garbage collection during linking. Previous commits allows each drivers and subsystems to specify which header files containing syscalls are relevant. So this commit changes the syscall generation to onl [... cut]
closed 2023-06-17 by nashif