zephyrproject-rtos/zephyr #58943: [Coverity CID: 248418] Out-of-bounds write in subsys/testsuite/ztest/src/ztress.c

zephyrbot, 2023-06-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/7b2034aaecc4cb2a261973b10b2fa608b29d398c/subsys/testsuite/ztest/src/ztress.c

Category: Memory - corruptions
Function: `ztress_execute`
Component: Testsuite
CID: [248418](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=248418)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/7b2034aaecc4cb2a261973b10b2fa608b29d398c/subsys/testsuite/ztest/src/ztress.c#L365

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

nashif, 2023-08-10:
false positive.

closed 2023-08-10 by nashif

fabiobaltieri, 2023-08-10:
@nashif could you look into https://github.com/zephyrproject-rtos/zephyr/issues/58677 as well? Andy was asking how to flag false positives, I think one needs permissions on the tool or something.