zephyrproject-rtos/zephyr #58971: [Coverity CID: 318608] Logically dead code in subsys/bluetooth/controller/ll_sw/ull_conn.c

zephyrbot, 2023-06-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/7b2034aaecc4cb2a261973b10b2fa608b29d398c/subsys/bluetooth/controller/ll_sw/ull_conn.c

Category: Control flow issues
Function: `ull_conn_done`
Component: Bluetooth
CID: [318608](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=318608)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/7b2034aaecc4cb2a261973b10b2fa608b29d398c/subsys/bluetooth/controller/ll_sw/ull_conn.c#L1010

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_DEADCODE). ([CWE-561](http://cwe.mitre.org/data/definitions/561.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cvinayak, 2023-06-30:
<html><body>
<h4><span ng-switch="" on="changeItem.fromCommit" class="ng-binding">2023-06-07 15:24:09<span ng-switch-default="" class="ng-binding ng-scope"> </span></span></h4>

Classification: | Unclassified | → | Bug
-- | -- | -- | --
Severity: | Unspecified | → | Minor
Action: | Undecided | → | Fix Required

</body>
</html>

fabiobaltieri, 2023-09-05:
@cvinayak could you look into this?

closed 2023-10-04 by carlescufi