zephyrproject-rtos/zephyr #59060: tests/boards/intel_adsp/hda_log fails on intel_adsp_cavs25

nashif, 2023-06-08:
**Describe the bug**
Various tests in `tests/boards/intel_adsp/hda_log` are failing when run on intel_adsp_cavs25 with no output on the screen

**To Reproduce**
Steps to reproduce the behavior:

`/scripts/twister -p intel_adsp_cavs25 --device-testing --hardware-map ~/map.yaml -T tests/boards/intel_adsp/hda_log`

**Expected behavior**
test should pass

The issue was introduced with 7d54586751cfe405d2cdcfd850f8100bc9844f63, which looks very related.

teburd, 2023-06-08:
Thanks for hunting down the source @nashif

tmleman, 2023-06-15:
What my change did is that IPC is not longer viewed as completed until we get acknowledgement from the host.

SOF has updated version of zephyr that contains this change. Both MTL and TGL work properly. I suppose the problem is that the test framework component that simulates the host does not handle incoming IPCs properly.

If I'm wrong I can propose fix that will revert this change and solve my initial problem different way.

nashif, 2023-06-15:
@teburd  based on the comment from @tmleman  can you please see if this needs to be dealt with on the test side?

teburd, 2023-06-27:
So after a few rounds of re-acquainting myself of all of this I believe the issue here is...

hda log wants to hold a spin lock while sending IPC

the spin lock prevents the ipc isr from running (interrupt masking)

We spin forever.

I have a branch which avoids IPC for hda log but to be honest no one is using hda log so I think some discussion needs to happen about what to do here.

PR #67564 (2024-01-17): logging: Remove hda log backend
The HDA log backend has been buggy for some time, and the fixes are well
understood. The issue with HDA log as it is implemented today means
there are concurrency and ISR issues where it can deadlock.

The fixes are effectively dropping the usage of ipc to communicate
status, using polling on the hda stream, and having a formatter per call
context (cores + isr) when log mode immediate is enabled to fix log
mangling. However it was found this had a lot of unintentional side
effects that made it difficult to make progress on when it came to
changes of the host sid [... cut]
closed 2024-01-17 by carlescufi