zephyrproject-rtos/zephyr #59068: Product hangs when it receives more ~2048 bytes HTTP packet

KalpeshKadam-eaton, 2023-06-08:
Board used : Nucleo_F429zi
Zephyr version : V3.2.0 (https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v3.2.0)

Problem statement : 
We have developed webserver with help of zephyr. When we send a HTTP packet to device ( PUT Method ) with length ~2048 then device (Nucleo_F429zi) hangs and soon watchdog reset is rebooting device.
If I disable watchdog then device do not reset and it does not ping back. Thus becomes unresponsive.

**To Reproduce**
Steps to reproduce the behavior:
1. Build project with watchdog enabled
2. Flash image
3. Device will start responding as static ip configured.
4. Use any rest client tool to send Small size (~512 chars) with method-PUT
5.  Check if device is in good state
6.  Use any rest client tool to send BIG size (~2048 chars) with method-PUT
7. Check if device is in good state

**Expected behavior**
After Step 5 and step 7, In both case device should be responsive.

**Actual behavior**
After step 5, Device remains in good state as I can ping the device.
But issue observed after step 7, when we send BIG HTTP request (~2048), Device will reboot due to watchdog reset. 
Device becomes unresponsive if watchdog is disabled while building project.

**Impact**
Instead of 2048 bytes, I have to use 512 bytes to make multiple WRITES successful. Overall WRITE is not meeting our benchmark.

**Attachment**

[Autoconf_device_hang_with_2048_char)HTTP_Request.docx](https://github.com/zephyrproject-rtos/zephyr/files/11688317/Autoconf_device_hang_with_2048_char.HTTP_Request.docx)

KalpeshKadam-eaton, 2023-06-08:
https://github.com/zephyrproject-rtos/zephyr/blob/4256cd41df6c60f1832fd2deb14edc30ac7debab/drivers/ethernet/eth_stm32_hal.c#L213

I found control gets locked in this while() loop

erwango, 2023-06-09:
Hi,

Starting release V3.2.0, we've been migrating STM32 Eth driver to the new Cube HAL Ethernet API, see https://github.com/zephyrproject-rtos/zephyr/issues/41380.
This work has been started first on H7 and then on F4/F7, it could be used by enabling `ETH_STM32_HAL_API_V2` Kconfig option.

This new API is fixing a number of issues that were usually seen with initial API, and probably the bug you're seeing.
Even if only fully available in V.3.3.0, I'd suggest to study how to use it, for instance by back porting on V3.2.0

KalpeshKadam-eaton, 2023-06-09:
@erwango  Thanks for quick response

I am referring https://github.com/zephyrproject-rtos/zephyr/pull/45770/files PR

erwango, 2023-06-12:
> Is there any specific section in PR which might help to solve my problem

Unfortunately not. The proposal is to use the new API, and this is the purpose of #45770. Using it partly (mix of new a old API), will likely end up in unexpected results.

So, on HAL you need to apply following commit:
https://github.com/zephyrproject-rtos/hal_stm32/commit/15993325e3aa961fbf5ffc17583ab1d4fed1e0bc
On F4, I expect merging this commit will not raise specific issues.

On zephyr repo, this is likely more tricky, but basically, adding all the 2023 commits in the link below should get you to a working and validated config:
https://github.com/zephyrproject-rtos/zephyr/commits/main/drivers/ethernet/eth_stm32_hal.c

closed 2023-08-27 by github-actions[bot]