zephyrproject-rtos/zephyr #59083: NRFConnect & Zephyr CMake Error: No SOURCES given to Zephyr library: drivers__disk

scath999, 2023-06-09:
**Bug Description**
SD Card access does not appear to be properly included in a build despite configuration settings. This is apparent from the CMake error

```
CMake Warning at C:/nordic/v2.3.0/zephyr/CMakeLists.txt:824 (message):
  No SOURCES given to Zephyr library: drivers__disk

  Excluding target from build.
```

I'm unsure where this error is occurring, since there are multiple points of possible failure. For example, I cannot `CONFIG_DISK_ACCESS_SDHC`, or `CONFIG_DISK_ACCESS_SPI_SDHC`  as they appear to be unknown symbols. However, in every example project that I look at, these flags appears to be enabled. 

This is compounded by the fact that Guiconfig/menuconfig appear to have a freehand overwriting various configuration options silently. This means I cannot be sure this isn't a configuration error arising out of an unrelated bug. However, these configuration options do not appear to change.

Here are the relevant lines from prj.conf
```
# FileIO
CONFIG_SPI=y
CONFIG_LOG=y
CONFIG_FILE_SYSTEM=y
CONFIG_DISK_ACCESS=y
CONFIG_DISK_DRIVERS=y
CONFIG_DISK_ACCESS=y

#These appear to be unknown symbols
# CONFIG_DISK_ACCESS_SDHC=y 
# CONFIG_DISK_ACCESS_SPI_SDHC=y
# CONFIG_DISK_SDHC_VOLUME_NAME="SD"

CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_FS_FATFS_EXFAT=y
CONFIG_FS_FATFS_LFN=y
CONFIG_FS_FATFS_MOUNT_MKFS=y
CONFIG_FS_FATFS_LFN_MODE_HEAP=y
```

Important cues/clues: Debugging `disk_access_get_di()` shows that the static `sys_dlist_t` named `disk_access_list` has a single entry. However, the `foreach` macro designed to loop over the list is never entered, causing this method to return `NULL`.

**Target Platform:** Laird MG100 with an nrf52840

**To Reproduce**
Steps to reproduce the behavior:
1. Set project configurations for SD Card access for an MG100 project as given above.
2. Include source for accessing the SD Card as in the `fat_fs` sample in the zephyr repository.
3. Build using WEST or NRFConnect's build tool.

**Expected Result:** Libraries should be properly included, allowing easy access to an attached SD Card.

**Actual Result:** The build succeeds with the CMake Warning given above. No SD Card access is possible.

**Impact**
This prevents us from saving important information to disk, endangering data integrity.

**Environment (please complete the following information):**
 - OS: Win10
 - Toolchain: NRFConnect v2.3
 - Zephyr 3.3.99

nordicjm, 2023-06-09:
You are not using zephyr and are using ncs, please report this issue to https://devzone.nordicsemi.com/ unless you are able to reproduce it directly with upstream zephyr itself (if so, reply and edit the original post with the information on zephyr)

scath999, 2023-06-09:
Thanks for the advice. Your wonderful interpersonal skills really shine here, jm.

nordicjm, 2023-06-12:
Thanks for not reading the bug description, which clearly states:
```
This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.
```

scath999, 2023-06-12:
Sigh. 

I tried, Jamie - sadly, Nordic doesn't have an issues tab on their github repo.  I invite you to investigate before typing further: https://github.com/nrfconnect/sdk-zephyr

I also looked on their "devzone" and found that that Nordic doesn't know any better than I do what the heck is going wrong, and hasn't for months - hence my post here. 

**You could have asked me questions about this rather than quickly and smugly dismissing me.** Doing so would have demonstrated maturity and insight on your part.

Instead, you've consistently chosen a condescending stance, which is both uncalled for an unprofessional. You've continued this attitude further with your last post. Yes, I called you out on it in a tongue-in-cheek way; Your response invited a remark on your behavior. Don't defend the mistake - take the note and move on. You represent your vendor - when you behave this way you make Nordic look bad.

I invite you to do better. Your attitude is helping no-one - you least of all.

jfischer-no, 2023-06-12:
Ohh, this is getting dramatic :smile_cat: 

@scath999 have you described SD card hardware in your devicetree?  There should be "zephyr,sdhc-spi-slot" node, like

https://github.com/zephyrproject-rtos/zephyr/blob/bfa0a87277c31d9e1b36104a29b17f3fc1a47444/boards/shields/adafruit_2_8_tft_touch_v2/dts/adafruit_2_8_tft_touch_v2.dtsi#L39-L48

Kconfig.spi would look for "zephyr,sdhc-spi-slot" compatible and enable CONFIG_SPI_SDHC and select CONFIG_SDHC_SUPPORTS_SPI_MODE.

https://github.com/zephyrproject-rtos/zephyr/blob/bfa0a87277c31d9e1b36104a29b17f3fc1a47444/drivers/sdhc/Kconfig.spi#L4-L7

Compatible "zephyr,sdmmc-disk"; would enable CONFIG_DISK_DRIVER_SDMMC
https://github.com/zephyrproject-rtos/zephyr/blob/bfa0a87277c31d9e1b36104a29b17f3fc1a47444/drivers/disk/Kconfig.sdmmc#L7-L12

scath999, 2023-06-13:
> Ohh, this is getting dramatic 😸

Hahah, true 😸 Ok, let's calm this all down a notch. Thanks for stepping in. 

> @scath999 have you described SD card hardware in your devicetree? There should be "zephyr,sdhc-spi-slot" node, like

I have, though the definition is different for the MG100. Note that I don't have an `spi0` or `spi1` definition for that board.

```
&spi2 {
	compatible = "nordic,nrf-spim";
	status = "okay";
	pinctrl-0 = <&spi2_default>;
	pinctrl-1 = <&spi2_sleep>;
	pinctrl-names = "default", "sleep";
	cs-gpios = <&gpio0 29 GPIO_ACTIVE_LOW>;
	sdhc0: sdhc@0 {
		compatible = "zephyr,mmc-spi-slot";
		reg = <0>;
		spi-max-frequency = <8000000>;
	};
};```

The `mmc` entry was suggested on Nordic's forums, and I've tried adding that, too.

```c
                ...
                sdhc0: sdhc@0 {
		compatible = "zephyr,mmc-spi-slot";
		reg = <0>;
		
		status = "okay";
		mmc {
		 	compatible = "zephyr,sdmmc-disk";
		 	status = "okay";
		};
		spi-max-frequency = <8000000>;
	};
```
 
> Kconfig.spi would look for "zephyr,sdhc-spi-slot" compatible and enable CONFIG_SPI_SDHC and select CONFIG_SDHC_SUPPORTS_SPI_MODE.

Interestingly, neither SPI hardware bus support nor SDHC drivers were enabled in my build. I didn't know I should

> Compatible "zephyr,sdmmc-disk"; would enable CONFIG_DISK_DRIVER_SDMMC

As JM rightly pointed out, I'm using the downstream version of Zephyr created by Nordic Semi. The SDMMC driver appears to be set by default, but it didn't work.

**Question:** 
While working inside `guiconfig`, I notice that the `SDMMC access via SD subsystem (NEW)` is disabled because of a dependency: `DT_HAS_ZEPHYR_SDMMC_DISK_ENABLED` must be `y`, but I'm not seeing a way to enable this.

Thoughts?

 
> See also https://docs.zephyrproject.org/latest/services/storage/disk/access.html#sd-card-support-via-spi

Thanks for sending this. I've looked at it before, but will review it again.

Thanks again for taking time :) Any guidance is welcome.

jfischer-no, 2023-06-13:
Okay, I just realized there is upstream support for MG100 board, and it looks like sdhc node uses outdated compatible `"zephyr,mmc-spi-slot"`. I changed it to "zephyr,sdhc-spi-slot" and added missing mmc, and built fat_fs sample.

```diff
diff --git a/boards/arm/mg100/mg100.dts b/boards/arm/mg100/mg100.dts
index 16e4c4a9fb..8d9d8c7f8c 100644
--- a/boards/arm/mg100/mg100.dts
+++ b/boards/arm/mg100/mg100.dts
@@ -130,9 +130,14 @@
        pinctrl-names = "default", "sleep";
        cs-gpios = <&gpio0 29 GPIO_ACTIVE_LOW>;
        sdhc0: sdhc@0 {
-               compatible = "zephyr,mmc-spi-slot";
+               compatible = "zephyr,sdhc-spi-slot";
                reg = <0>;
+               status = "okay";
                spi-max-frequency = <8000000>;
+               mmc {
+                       compatible = "zephyr,sdmmc-disk";
+                       status = "okay";
+               };
        };
 };
```

`west build -b mg100 samples/subsys/fs/fat_fs -- -DCONFIG_DISK_DRIVER_SDMMC=y -DCONFIG_SPI=y`

It seems to work.

```shell
foo@bar:~/zephyr-project/zephyr$ rgrep DISK_DRIVER build/zephyr/.config 
CONFIG_DISK_DRIVER_SDMMC=y
CONFIG_DISK_DRIVERS=y
# CONFIG_DISK_DRIVER_RAM is not set
# CONFIG_DISK_DRIVER_FLASH is not set
# CONFIG_DISK_DRIVER_MMC is not set
foo@bar:~/zephyr-project/zephyr$ rgrep SDMMC build/zephyr/.config 
CONFIG_DISK_DRIVER_SDMMC=y
CONFIG_DT_HAS_ZEPHYR_SDMMC_DISK_ENABLED=y
CONFIG_SDMMC_VOLUME_NAME="SD"
CONFIG_SDMMC_SUBSYS=y
# CONFIG_SDMMC_LOG_LEVEL_OFF is not set
# CONFIG_SDMMC_LOG_LEVEL_ERR is not set
# CONFIG_SDMMC_LOG_LEVEL_WRN is not set
# CONFIG_SDMMC_LOG_LEVEL_INF is not set
# CONFIG_SDMMC_LOG_LEVEL_DBG is not set
CONFIG_SDMMC_LOG_LEVEL_DEFAULT=y
CONFIG_SDMMC_LOG_LEVEL=3
CONFIG_SDMMC_STACK=y
```

This looks like a bug in boards/arm/mg100/mg100.dts and can be reopened, FYI @rerickson1.

jfischer-no, 2023-06-13:
Reopening because of:

https://github.com/zephyrproject-rtos/zephyr/blob/3104ad0e393cc4df756b2d12aa15c0b3c9be3960/doc/releases/release-notes-3.1.rst?plain=1#L455-L459

https://github.com/zephyrproject-rtos/zephyr/blob/3104ad0e393cc4df756b2d12aa15c0b3c9be3960/doc/releases/release-notes-3.1.rst?plain=1#L911-L918

There are two outdated compatible,
```shell
foo@bar:~/zephyr-project/zephyr$ git grep "zephyr,mmc-spi-slot"
boards/arm/mg100/mg100.dts:             compatible = "zephyr,mmc-spi-slot";
doc/releases/release-notes-3.1.rst:    * ``zephyr,mmc-spi-slot``
tests/drivers/build_all/sensor/spi.dtsi:        compatible = "zephyr,mmc-spi-slot";
```

scath999, 2023-06-13:
@jfischer-no Thank you so much. I've also built this successfully. I do get some odd feedback from the console when building (below), but it doesn't seem like a showstopper.

I now want to integrate this with my application. However, your solution appears to depend on parts of zephyr that aren't present/configurable in Nordic's fork. 

I think I can alleviate this by updating their Zephyr with `west` - unsure if this will break functionality in their version of Zephyr, but seems like a good idea to try.

Thoughts?

Many thanks, again.

```
*** Booting Zephyr OS build zephyr-v3.3.0-1195-g9a759025d912 ***
[00:00:00.356,781] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.357,574] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.358,337] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.359,100] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.359,832] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.360,595] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.361,358] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.362,121] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.362,884] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.363,647] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.364,379] <inf> sd: Legacy card detected, no CMD8 support
[00:00:00.364,410] <inf> sd: Card does not support CMD8, assuming legacy card
[00:00:06.406,860] <inf> sd: Maximum SD clock is under 25MHz, using clock of 8000000Hz
```

scath999, 2023-06-14:
Update:

A couple of new issues have arisen when trying to build my project.

Having updated my local copy of Nordic's fork of zephyr, I get the following error when trying to build using NRFConnect:

```
error: DT_HAS_ZEPHYR_SDMMC_DISK_ENABLED (defined at D:/Ginkgo/Development/code/nrf_zeph_test/nrf_zep
hyr/samples/net/sockets/_ginkgo_hub/build/Kconfig/Kconfig.dts:5917) is assigned in a configuration
file, but is not directly user-configurable (has no prompt). It gets its value indirectly from other
symbols. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_DT_HAS_ZEPHYR_SDMMC_DISK_ENABLED and/or
look up DT_HAS_ZEPHYR_SDMMC_DISK_ENABLED in the menuconfig/guiconfig interface. The Application
Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of
the manual might be helpful too.
```

The suggested web link takes me to the Kconfig Search with 0 items in it (opps?). This isn't alleviated when changing `/latest/` to `/3.3.0/`.

Aside: This is very frustrating - I've had multiple issues with Zephyr's configuration strategy, including configuration options being mysteriously and silently overwritten, causing my builds to fail for no apparent reason. A different ticket, but wow!

Trying to build the project manually using the west command you used results in another error:
``` 
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V] <command> ...
west: error: argument <command>: invalid choice: 'build' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 'forall', 'help', 'config', 'topdir', 'selfupdate')
```

If I then initialize west using `west init -l c:\nordic\v2.3.0\zephyr`, I'm told to run `west update` from `c:\nordic\zephyr`. 

Once I do that, I return to the build directory and try to build using west again, at which point I'm given the `invalid choice: 'build'` message again.

Unsure how to proceed. I know this solution works in isolation - integrating it with my work should be relatively easy.

Thoughts?

closed 2023-06-19 by carlescufi