zephyrproject-rtos/zephyr #59087: stm32h7: qspi: Not acessible in memory mapped mode

erwango, 2023-06-09:
**Describe the bug**
Issue raised from a coworker following merge of https://github.com/zephyrproject-rtos/zephyr/pull/57467:

> I'm not sure if this is the best solution, although I have zero experience with Zephyr RTOS (I have some basic knowledge of the DTS concept from Linux).
As I understand, the code will disable (Strongly ordered, no access) the QSPI memory region by default for all Zephyr projects. Then if you want to use QSPI in memory-mapped mode in your project, you need to override this setting in the project specific DTS.

> I think better solution would be to have Region 0 with following configuration:
> - Base address: 0x0
> - Region size: 4GB (whole memory range)
> - SubRegion disable: 0x87 (this excludes internal memories and peripherals)
> - TEX: 0, Bufferable: 0, Cacheable: 0, Shareable: 1 (Strongly ordered)
> - All access disabled
> - Instruction access: disabled

> Since this is defined as Region 0, any higher MPU regions can overwrite such setting.
This what is recommended and applied via pop-up window when configuring new CubeMX project for STM32F7 or STM32H7.

> This solution might prevent similar issue in more wide range of projects. E.g. similar issues can be encountered with FMC peripheral.
However I'm not sure how easy it would be to integrate to Zephyr, since I haven't seen e.g. the subregion option for the zephyr,memory-region node. Also I'm not sure if it possible to force this configuration to be region 0. Maybe it would need to be done some other way.

> Also I'm not sure if "EXTMEM" is a good name for attribute that disables all access.

**Impact**
the code will disable (Strongly ordered, no access) the QSPI memory region by default for all Zephyr projects. Then if you want to use QSPI in memory-mapped mode in your project, you need to override this setting in the project specific DTS.

erwango, 2023-07-05:
@benjaminbjornsson ^^, This point will have to be solved (or commit reverted) before being able to enable XIP on H7 series.

erwango, 2023-07-07:
@carlocaione I'm staring at https://github.com/zephyrproject-rtos/zephyr/pull/60049 right now. 
Do you think it could solve current point?
My understanding is not fully, but can now be extended to fulfill it. Am I right ?

carlocaione, 2023-07-07:
That commit is only introducing a couple of helpers to manage memory nodes and their attributes, what you do with these helpers is really up to the developer. So yes, you can leverage that commit if you want.

I think what you specifically need from that commit is the fact that you can use a memory attribute on an `mmio-sram` region and build an MPU region on that, so yeah, I think that could be helpful for this case.

erwango, 2023-07-07:
Thanks @carlocaione we'll have a closer look

FRASTM, 2023-07-26:
Adding such a _region 0_  to the static const struct arm_mpu_region mpu_regions[]  in /soc/arm/st_stm32/stm32h7/mpu_regions.c   with 
```
	MPU_REGION_ENTRY("REGION_0", 0,
					{ (STRONGLY_ORDERED_SHAREABLE | REGION_4G | NO_ACCESS_Msk) }),

```
is not enough

FRASTM, 2023-08-01:
When declaring the external qspi NOR flash node with
```
	quadspi_memory: memory@90000000 {
		compatible = "zephyr,memory-region", "mmio-sram";
		reg = <0x90000000 DT_SIZE_M(256)>;
		zephyr,memory-region = "QSPI";
		zephyr,memory-attr = "EXTMEM";
	};
```
The application (gdb neither)  cannot not access this memory location 
```
gdb) x /x 0x90000000
0x90000000:	0x00000000
(gdb) Error: jtag status contains invalid mode value - communication failure
Polling target stm32h7x.cpu0 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
```

FRASTM, 2023-08-07:
See PR #61122

FRASTM, 2023-08-07:
When Using the qspi NOR flash in MemoryMapped mode, it is accessible for reading
Even is writing is possible it is not recommended and the erase is is not possible.

--> There is a question about what should be driver behavior on flash_erase/flash_write API in MemoryMapped mode.

FRASTM, 2023-10-25:
For the stm32H7x serie, when the zephyr driver is executing the _same_ sequence as the stm32CubeMX,  to enter the MemoryMapped mode, then the memory at 0x90000000 becomes readable (write always fails) on half of the the byte address :
offset = 00000000 ; read 20 
offset = 00000001 ; read 2a 
offset = 00000002 ; read 2a 
offset = 00000003 ; read 65 
offset = 00000004 ; read 6f 
offset = 00000005 ; read 79 
offset = 00000006 ; read 6d 
offset = 00000007 ; read 70 
offset = 00000008 ; read 65 

[image]

closed 2024-01-08 by github-actions[bot]