zephyrproject-rtos/zephyr #59096: Wrong partitioning in nucleo_f410rb devicetree

nguyenmthien, 2023-06-09:
**Describe the bug**
The board nucleo_f410rb apparently have a erroneous devicetree definition file. The devicetree partitioned the chip to [four differnt partitions with total size of 512KB](https://github.com/zephyrproject-rtos/zephyr/blob/a8ed28ab6fc868ac73eed56fc3b46c23df52adde/boards/arm/nucleo_f410rb/nucleo_f410rb.dts#L121) although the [flash size is only 128KB](https://www.st.com/en/microcontrollers-microprocessors/stm32f410rb.html).

**Impact**
This will make some firmware e.g. mcuboot unable to be flashed into the device.

erwango, 2023-06-09:
@nguyenmthien Fell free to update the partition to something more reasonable (or purely remove it)

FRASTM, 2023-07-31:
Would it be better to reduce the boot_partition to 32KB, and use the 128KB in 4 partitions of 32K each 

```
/*
		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(32)>;
			read-only;
		};

		slot0_partition: partition@8000 {
			label = "image-0";
			reg = <0x00008000 DT_SIZE_K(32)>;
		};
		slot1_partition: partition@10000 {
			label = "image-1";
			reg = <0x00010000 DT_SIZE_K(32)>;
		};
		scratch_partition: partition@18000 {
			label = "image-scratch";
			reg = <0x000018000 DT_SIZE_K(32)>;
		};

```

FRASTM, 2023-07-31:
[image]

nguyenmthien, 2023-07-31:
> Would it be better to reduce the boot_partition to 32KB, and use the 128KB in 4 partitions of 32K each
>
> ```
> 		boot_partition: partition@0 {
> 			label = "mcuboot";
> 			reg = <0x00000000 [...]

Hi, for MCUBoot, 32KB of Flash memory is a bit strict. I suggest having a single-slot configuration with 64KB for MCUBoot and the rest for the image

FRASTM, 2023-07-31:
> 
> Hi, for MCUBoot, 32KB of Flash memory is a bit strict. I suggest having a single-slot configuration with 64KB for MCUBoot and the rest for the image

Yes, however some boards have boot_partition of 32KB.  
It is still possible to have a boot_partition of 48K and reduce the scratch_partition to 16KB
and several other combinations are still valid.

closed 2023-08-30 by carlescufi