zephyrproject-rtos/zephyr #59100: boards: arc: emsdp: init code used for peripheral-specific operations

gmarull, 2023-06-09:
**Describe the bug**

It looks like the `emsdp` ARC board has some custom init code to set peripheral-specific registers. It seems to hardcode peripheral addresses and enable their clock unconditionally in a `SYS_INIT` call defined at board level.

https://github.com/zephyrproject-rtos/zephyr/blob/a8ed28ab6fc868ac73eed56fc3b46c23df52adde/boards/arc/emsdp/platform.c#L10-L23

**To Reproduce**

Open https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arc/emsdp/platform.c

**Expected behavior**

It is driver's responsability to handle this, not board. Hardcoding addresses, or running this code regardless of SPI device states is also wrong.

**Impact**

This is not how driver-specific actions should be done in Zephyr.

**Logs and console output**

N/A

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.1
 - Commit SHA or Version used a8ed28ab6fc868ac73eed56fc3b46c23df52adde

**Additional context**

Discovered as part of SYS_INIT usage cleanup in boards

evgeniy-paltsev, 2023-06-09:
As I remember (@SiyuanCheng-CN please correct me if I'm wrong) it's platform specific part maped into driver address-space.
So it would differ from platform to platform.

gmarull, 2023-06-09:
> As I remember (@SiyuanCheng-CN please correct me if I'm wrong) it's platform specific part maped into driver address-space. So it would differ from platform to platform.

Note that generic IP drivers can also have vendor-specific quirks in a separate file if needed.

SiyuanCheng-CN, 2023-06-09:
> As I remember (@SiyuanCheng-CN please correct me if I'm wrong) it's platform specific part maped into driver address-space. So it would differ from platform to platform.

Yes it is EMSDP specific.

closed 2023-10-23 by github-actions[bot]