zephyrproject-rtos/zephyr #59114: Out Of Tree Board Fails with Sysbuild

JeremiahGillis, 2023-06-09:
**Describe the bug**
Project fails to build using `Sysbuild` with an out of tree board.

This issue was first posted #53780 and fixed with #56801. The issue appeared again at aff9683387ff3fc6a88cb1a507c562fd44359bf3 and is still present as of 6e940c513cb665a70fb57c2dd409d0bbdba734f2.

Please also mention any information which could help others to understand the problem you're facing:
 - nrf52840
 - Issue can be bypassed by using any commit from cd9465ac940ca9fdc66a1d95c158a31a06e471cb to f4cbb0107476b533af87ca2e4c53464aaac040d2
 - This occurs with my application and custom out of tree board. I can reproduce this issue with Zephyr examples.

**To Reproduce**
Steps to reproduce the behavior:
1. git checkout main
2. To force out of tree example, delete `nrf52840dk_nrf52840` in `zephyr\boards\arm`. Otherwise Zephyr will build successful even though the out of tree board is specified.
3. west build -b nrf52840dk_nrf52840 -p always --sysbuild C:\Users\jerem\zephyrproject\zephyr\samples\hello_world -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DBOARD_ROOT=C:\Users\jerem\zephyrproject\zephyr\samples\application_development\out_of_tree_board
4. See error

**Expected behavior**
The build is successful. This should occur whether a board revision is specified or not.

**Impact**
This prevents mcuboot and application for being built and flashed to board.

**Logs and console output**
```
(.venv) C:\Users\jerem\zephyrproject\zephyr>west build -b nrf52840dk_nrf52840 -p auto --sysbuild C:\Users\jerem\zephyrproject\zephyr\samples\hello_world -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DBOARD_ROOT=C:\Users\jerem\zephyrproject\zephyr\samples\application_development\out_of_tree_board
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base): extensions sysbuild_extensions python west root zephyr_module boards shields sysbuild_kconfig
-- Found Python3: C:/Users/jerem/zephyrproject/.venv/Scripts/python.exe (found suitable exact version "3.11.0") found components: Interpreter
-- Cache files will be written to: C:/Users/jerem/zephyrproject/zephyr/.cache
-- Found west (found suitable version "0.14.0", minimum required is "0.14.0")
-- Board: nrf52840dk_nrf52840
Parsing C:/Users/jerem/zephyrproject/zephyr/share/sysbuild/Kconfig
Loaded configuration 'C:/Users/jerem/zephyrproject/zephyr/build/empty.conf'
Merged configuration 'C:/Users/jerem/zephyrproject/zephyr/build/empty.conf'
Merged configuration 'C:/Users/jerem/zephyrproject/zephyr/build/zephyr/misc/generated/extra_kconfig_options.conf'
Merged configuration 'C:/Users/jerem/zephyrproject/zephyr/build/empty.conf'
Configuration saved to 'C:/Users/jerem/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Users/jerem/zephyrproject/zephyr/build/autoconf.h'
--
   *****************************
   * Running CMake for mcuboot *
   *****************************

Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/jerem/zephyrproject/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.22.2
-- Found Python3: C:/Users/jerem/zephyrproject/.venv/Scripts/python.exe (found suitable exact version "3.11.0") found components: Interpreter
-- Cache files will be written to: C:/Users/jerem/zephyrproject/zephyr/.cache
-- Zephyr version: 3.4.0-rc2 (C:/Users/jerem/zephyrproject/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.14.0")
-- Board: nrf52840dk_nrf52840
No board named 'nrf52840dk_nrf52840' found.

Please choose one of the following boards:

arc:
  em_starterkit
  em_starterkit_em11d
  em_starterkit_em7d
  em_starterkit_em7d_v22
  emsdp
  emsdp_em4
  emsdp_em5d
  emsdp_em6
  emsdp_em7d
  emsdp_em7d_esp
  emsdp_em9d
  hsdk
  hsdk4xd
[... 556 lines cut ...]
  nxp_adsp_imx8m
  nxp_adsp_imx8x
  odroid_go
  olimex_esp32_evb
  qemu_xtensa
  xt-sim
-- Configuring incomplete, errors occurred!
CMake Error at cmake/modules/sysbuild_extensions.cmake:376 (message):
  CMake configure failed for Zephyr project: mcuboot

  Location: C:/Users/jerem/zephyrproject/bootloader/mcuboot/boot/zephyr/
Call Stack (most recent call first):
  CMakeLists.txt:90 (ExternalZephyrProject_Cmake)

-- Configuring incomplete, errors occurred!
See also "C:/Users/jerem/zephyrproject/zephyr/build/CMakeFiles/CMakeOutput.log".
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' -DWEST_PYTHON=C:/Users/jerem/zephyrproject/.venv/Scripts/python.exe '-BC:\Users\jerem\zephyrproject\zephyr\build' -GNinja -DBOARD=nrf52840dk_nrf52840 -DSB_CONFIG_BOOTLOADER_MCUBOOT=y '-DBOARD_ROOT=C:\Users\jerem\zephyrproject\zephyr\samples\application_development\out_of_tree_board' '-SC:\Users\jerem\zephyrproject\zephyr\share\sysbuild' '-DAPP_DIR:PATH=C:\Users\jerem\zephyrproject\zephyr\samples\hello_world'
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK 0.16.1
 - Commit 6e940c513cb665a70fb57c2dd409d0bbdba734f2

**Additional context**

57300, 2023-06-11:
Hi, thanks for catching this and finding the bad commit. Fix is here: https://github.com/zephyrproject-rtos/zephyr/pull/59130

closed 2023-06-12 by nashif

JeremiahGillis, 2023-06-12:
This fixed my issue. Thanks!