zephyrproject-rtos/zephyr #59116: Flash partition coredump backend broken when CONFIG_ASSERT=y

dennisgr102, 2023-06-09:
Description:

The flash partition coredump backend attempts to acquire a semaphore via k_sem_take() in interrupt context. When CONFIG_ASSERT=y k_sem_take() will assert. Given coredumps are specifically designed to run in interrupt context I believe the k_sem_take() should be removed or it should only be called if k_is_in_isr() returns false meaning the system is in thread context.

Steps to reproduce the behavior:

1. Build any board/app combination with CONFIG_ASSERT=y
2. Force a coredump via k_oops()
3. Observe assert (and lack of coredump)

Expected behavior:

No assert and coredump collection

Impact:

Impossible to collect a coredump via the flash partition backend when CONFIG_ASSERT=y

Console output:

ASSERTION FAIL [((arch_is_in_isr() == 0) || ((timeout).ticks == (((k_timeout_t) {0})).ticks))] @ ZEPHYR_BASE/kernel/sem.c:121

ubieda, 2023-06-14:
I came across this issue, which I've experienced as well. To solve it I've applied the attached patch to Zephyr. LMK if you think this is a good fit; if so, I can file a PR.

[0003-fix-Omit-k_sem_take-blocking-assertion-if-we-re-on-a.patch](https://github.com/zephyrproject-rtos/zephyr/files/11749733/0003-fix-Omit-k_sem_take-blocking-assertion-if-we-re-on-a.patch)

mrkhldn, 2023-06-14:
Would it be better to just update the flash backend to not take/give the semaphore in coredump_flash_backend_start/end? Nothing else should be running that could interfere at that point, right?

cc @dcpleung

ubieda, 2023-06-14:
> Would it be better to just update the flash backend to not take/give the semaphore in coredump_flash_backend_start/end? Nothing else should be running that could interfere at that point, right?

@mrkhldn I don't think that'll work when the underlying flash APIs use blocking calls (e.g: using an external Flash hooked up over SPI).

mrkhldn, 2023-06-14:
> @mrkhldn I don't think that'll work when the underlying flash APIs use blocking calls (e.g: using an external Flash hooked up over SPI).

I'm still not sure why taking a semaphore would be required though. Once you're in coredump_flash_backend_start, there won't be other threads trying to simultaneously access the flash, only the coredump handler code will run until its completed.

ubieda, 2023-06-14:
@mrkhldn Right - I do agree removing the semaphore from `coredump_flash_backend` will address the issue when the fault occurs. I imagine the semaphore also attempts to provide some sort of thread-safety for processing the coredump: reading, erasing, etc.

mrkhldn, 2023-06-14:
Right, so I guess my suggestion would be to update partition_open() to take a param indicating whether or not to skip the semaphore. Then only use the semaphore for the read/erase operations, not the actual dump writing while handling a crash. This approach seems more tactical than updating kernel code.

mrkhldn, 2023-06-15:
Ah, I'm not sure what the right approach would be there. @tbursztyka is it safe to access the semaphore in the spi driver mentioned above during a crash dump? Should we be looking for some way to disable the assert causing problems?

dennisgr102, 2023-06-15:
I think the fundamental question here is: Are coredumps ever taken in thread context?

If the answer is no then removing the sem_take/sem_give is the right approach given they are not working as it currently is. It's just that the error is not seen unless CONFIG_ASSERT is on. In other words, by removing the semaphore usage we'd be in no different of a scenario then we are now.

If the answer is yes the solution becomes a little more complicated. For example, we may want to think about using k_is_in_isr() to determine if the semaphore should be used.

tbursztyka, 2023-07-11:
The problem description is a bit cryptic, which semaphore are we talking about?  spi context's one or partition_open or any semaphore on the way to handle a coredump? 

I don't know this subsystem, but being in isr mode, dealing with a SPI based flash, this could collide with an on-going SPI transaction. So you could force your transaction to take over, but this would cascade into another issue: the primary  SPI transaction caller would end up in an unknown state (hang up, crash or?). Actually this is not limited to SPI, and seems more like a generic problem.

I really should take a look at this coredump part, because a lot of questions are popping up in my head such as: : 
- What about coredumped thread owned kernel objects?
- What if the coredumped thread was in the middle of something affecting the coredump handling itself (flash, spi, you name it...)?*
- etc...

*This reminds me, that I proposed long ago to add a way to cancel any device call, whatever it would be. Seems like this use-case exposed here would be a relevant user of such feature.

jhedberg, 2023-09-12:
What's the status of this issue? Someone still working on it?

jhedberg, 2023-09-12:
@tbursztyka @mrkhldn (see previous comment)

closed 2023-11-26 by github-actions[bot]