zephyrproject-rtos/zephyr #59128: subsys: display: glitchy display in lvgl demo for stm32f746g_disco

ric96, 2023-06-11:
I just built the subsys/display/lvgl demo for the stm32f746g_disco board and the screen periodically glitches. I does not seem to be related to the main loops sleep duration, i tried playing around with it.

Video:

https://github.com/zephyrproject-rtos/zephyr/assets/5900740/704c561d-7cf8-4716-ac72-d44c133adbca

ric96, 2023-06-13:
It does not seem to be related to lvgl, i tried the `driver/display` sample as well with the same issue.
I have tried copying over the display clock values from the STM32Cube sample application `LTDC_Display_1Layer` but the issue persists.

```
diff --git a/boards/arm/stm32f746g_disco/stm32f746g_disco.dts b/boards/arm/stm32f746g_disco/stm32f746g_disco.dts
index 93054659df..3f9266448e 100644
--- a/boards/arm/stm32f746g_disco/stm32f746g_disco.dts
+++ b/boards/arm/stm32f746g_disco/stm32f746g_disco.dts
@@ -273,12 +273,13 @@ zephyr_udc0: &usbotg_fs {
                pixelclk-active = <0>;
                hsync-active = <0>;
                vsync-active = <0>;
-               hsync-len = <1>;
+               hsync-len = <41>;
                vsync-len = <10>;
-               hback-porch = <43>;
-               vback-porch = <12>;
-               hfront-porch = <8>;
-               vfront-porch = <4>;
+               hback-porch = <13>;
+               vback-porch = <2>;
+               hfront-porch = <32>;
+               vfront-porch = <2>;
+               clock-frequency = <9500000>;
        };
        def-back-color-red = <0xFF>;
        def-back-color-green = <0xFF>;
```

dolence, 2023-12-07:
Hi! 
I'm having the same issue on STM32H7. Did you managed to solve this?

kartben, 2023-12-13:
Can confirm that F746G Disco is glitchy for me as well.

dolence, 2023-12-13:
@erwango would be possible to reopen this? 3 persons with the same problem.

coskunergan, 2023-12-14:
using 800x480px with F746G Disco and I have the same problem. Looks like this issue needs a solution.

kartben, 2023-12-14:
Maybe #66513 will help?

kartben, 2023-12-14:
> @erwango would be possible to reopen this? 3 persons with the same problem.

Done

erwango, 2023-12-14:
@ajarmouni-st Can you have a look ?

dolence, 2023-12-14:
> Maybe #66513 will help?

I'm very new to Zephyr and my git knowledge resumes to cloning repositories. How could I test this PR?

ajarmouni-st, 2023-12-14:
@dolence https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/checking-out-pull-requests-locally

dolence, 2023-12-14:
I have tried #66513 and while the double framebuffer was moved to external SDRAM the flickering becomes worst and the CPU usage has increased.

```
Memory region         Used Size  Region Size  %age Used
           FLASH:      287508 B         1 MB     27.42%
             RAM:       38040 B       512 KB      7.26%
            QSPI:          0 GB       256 MB      0.00%
           SRAM1:          0 GB       128 KB      0.00%
           SRAM2:          0 GB       128 KB      0.00%
           SRAM3:          0 GB        32 KB      0.00%
           SRAM4:          0 GB        64 KB      0.00%
          SDRAM2:       4000 KB         8 MB     48.83%
        IDT_LIST:          0 GB         2 KB      0.00%
```

https://github.com/zephyrproject-rtos/zephyr/assets/5223003/3b6a2884-a7b0-490c-9ce1-5a7a85723dd0

ric96, 2023-12-16:
> I have tried #66513 and while the double framebuffer was moved to external SDRAM the flickering becomes worst and the CPU usage has increased.
> 
> ```
> Memory region         Used Size  Region Size  %age Used
>            FLASH:      287508 B         1 MB     27.42%
>              RAM:       38040 B       512 KB      7.26%
>             QSPI:          0 GB       256 MB      0.00%
>            SRAM1:          0 GB       128 KB      0.00%
>            SRAM2:          0 GB       128 KB      0.00%
>            SRAM3:          0 GB        32 KB      0.00%
>            SRAM4:          0 GB        64 KB      0.00%
>           SDRAM2:       4000 KB         8 MB     48.83%
>         IDT_LIST:          0 GB         2 KB      0.00%
> ```
> 
> flickering.mp4

you need to remove `zephyr,memory-attr` from devicetree

dolence, 2023-12-17:
> you need to remove `zephyr,memory-attr` from devicetree
Thank you. That solved the flickering. The CPU usage is around 34%, with 10 FPS. Maybe if enabling DMA2D this could be better. Also, text are uglier than before, as you described.

cameled, 2023-12-17:
> > you need to remove `zephyr,memory-attr` from devicetree
> 
> Thank you. That solved the flickering. The CPU usage is around 34%, with 10 FPS. Maybe if enabling DMA2D this could be better. Also, text are uglier than before, as you described.

You can try it with below config:
```
CONFIG_LV_USE_GPU_STM32_DMA2D=y
CONFIG_LV_GPU_DMA2D_CMSIS_INCLUDE="stm32h747xx.h"
```

dolence, 2023-12-17:
> > > you need to remove `zephyr,memory-attr` from devicetree
> > 
> > 
> > Thank you. That solved the flickering. The CPU usage is around 34%, with 10 FPS. Maybe if enabling DMA2D this could be better. Also, text are uglier than before, as you described.
> 
> You can try it with below config:
> 
> ```
> # PATH: modules/lib/gui/lvgl/Kconfig
> CONFIG_LV_USE_GPU_STM32_DMA2D=y
> # Please update 'stm32h747xx.h`.
> CONFIG_LV_GPU_DMA2D_CMSIS_INCLUDE="stm32h747xx.h"
> ```

I have tried this before but many warnings was being shown:

```
/home/dolence/zephyrproject/modules/lib/gui/lvgl/src/core/../draw/stm32_dma2d/lv_gpu_stm32_dma2d.h:56:13: warning: 'lv_draw_stm32_dma2d_blend' declared 'static' but never defined [-Wunused-function]
   56 | static void lv_draw_stm32_dma2d_blend(lv_draw_ctx_t * draw_ctx, const lv_draw_sw_blend_dsc_t * dsc);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/dolence/zephyrproject/modules/lib/gui/lvgl/src/core/../draw/stm32_dma2d/lv_gpu_stm32_dma2d.h:57:13: warning: 'lv_draw_stm32_dma2d_buffer_copy' declared 'static' but never defined [-Wunused-function]
   57 | static void lv_draw_stm32_dma2d_buffer_copy(lv_draw_ctx_t * draw_ctx,
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dolence/zephyrproject/modules/lib/gui/lvgl/src/core/../draw/stm32_dma2d/lv_gpu_stm32_dma2d.h:60:17: warning: 'lv_draw_stm32_dma2d_img' declared 'static' but never defined [-Wunused-function]
   60 | static lv_res_t lv_draw_stm32_dma2d_img(lv_draw_ctx_t * draw_ctx, const lv_draw_img_dsc_t * img_dsc,
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/home/dolence/zephyrproject/modules/lib/gui/lvgl/src/core/../draw/stm32_dma2d/lv_gpu_stm32_dma2d.h:62:13: warning: 'lv_gpu_stm32_dma2d_wait_cb' declared 'static' but never defined [-Wunused-function]
   62 | static void lv_gpu_stm32_dma2d_wait_cb(lv_draw_ctx_t * draw_ctx);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dolence/zephyrproject/modules/lib/gui/lvgl/src/core/../draw/stm32_dma2d/lv_gpu_stm32_dma2d.h:63:13: warning: 'lv_draw_stm32_dma2d_img_decoded' declared 'static' but never defined [-Wunused-function]
   63 | static void lv_draw_stm32_dma2d_img_decoded(lv_draw_ctx_t * draw_ctx, const lv_draw_img_dsc_t * img_dsc,
```

cameled, 2023-12-17:
> I have tried this before but many warnings was being shown:
>
> ```
> /home/dolence/zephyrproject/modules/lib/gui/lvgl/src/core/../draw/stm32_dma2d/lv_gpu_stm32_dma2d.h:56:13: warning: [...]

Same to me. I found there no change with this config.

dolence, 2023-12-17:
Probably some bug in DMA2D activation?

ric96, 2023-12-17:
> Probably some bug in DMA2D activation?

better to open a separate issue regarding this.

cameled, 2023-12-17:
> Probably some bug in DMA2D activation?

Looks `modules/lib/gui/lvgl/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c` write dma2d register directly.

kartben, 2023-12-17:
I doubt any of the "GPU" accelerations available upstream are currently supported, and they probably should be hidden from Zephyr's Kconfig? /cc @faxe1008

faxe1008, 2024-01-05:
> I doubt any of the "GPU" accelerations available upstream are currently supported, and they probably should be hidden from Zephyr's Kconfig? /cc @faxe1008 

With a bit of patching you can make this work. PXP support should be relatively trivial [https://github.com/lvgl/lvgl/blob/master/src%2Fdraw%2Fnxp%2Fpxp%2Flv_pxp_osa.c](https://github.com/lvgl/lvgl/blob/master/src%2Fdraw%2Fnxp%2Fpxp%2Flv_pxp_osa.c). The only plugging point missing really is a semaphore impl. Since v9 will come with an osal that already provides mutex abstraction maybe adding a semaphore in there is a cheap way to unlock those drivers?

Will take a stab on this next week. Thanks for bumping me into this.

kartben, 2024-01-05:
> Will take a stab on this next week. Thanks for bumping me into this.

w00t, awesome! And LVGL9 will make things even easier for us in this area, if I understand correctly?

hanh-nn, 2024-01-08:
Hi!
I have the same problem and removed ```zephyr,memory-attr``` but it didn't work for me. I describe my problem here #66993. 
Did I miss anything?

faxe1008, 2024-01-09:
@hanh-nn does this PR solve your issue: [https://github.com/zephyrproject-rtos/zephyr/pull/66513](https://github.com/zephyrproject-rtos/zephyr/pull/66513)?

hanh-nn, 2024-01-10:
> @hanh-nn PR này có giải quyết được vấn đề của bạn không: #66513?

My mistake. It's worked. Thanks

erwango, 2024-01-12:
@ric96 Can you confirm this is solved by https://github.com/zephyrproject-rtos/zephyr/pull/66513 ?

closed 2024-02-02 by ajarmouni-st