zephyrproject-rtos/zephyr #59161: LSE enable not working for STM32F1x chips

bdesterBE, 2023-06-12:
**Describe the bug**
When attempting to enable to the LSE for use with an STM32F105RCT6, application code with hang on the following line found in `clock_stm32_ll_common.c`. 
```
while (!LL_PWR_IsEnabledBkUpAccess()) {
    /* Wait for Backup domain access */
}
```
This appears to be the same behavior observed in #56449. I am just assuming that this issue would occur for all F1 series chips.

It may also be worth noting here that I'm guessing LSE has not been fully tested on F1 series chips yet, as its `&clk_lse` devicetree node found in `stm32f1.dtsi` does not have `st,stm32-lse-clock` set for its `compatible` property (and also lacks the required `driving-capability` property therefore).
```
clk_lse: clk-lse {
	#clock-cells = <0>;
	compatible = "fixed-clock";
	clock-frequency = <32768>;
	status = "disabled";
};
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build a sample project with `status = "okay";` added as a property in an STM32 F1 series &clk_lse devicetree node

**Environment**
OS: macOS
Toolchain: zephyr-sdk-0.15.2
Zephyr Version used: v3.2.99-ncs2

FRASTM, 2023-06-16:
It appears that void config_enable_default_clocks(void)  function of the [drivers/clock_control/clock_stm32f1.c ](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/clock_control/clock_stm32f1.c#LL159C13-L159C13) does 
enables the PWR domain clock 
( zephyr version v3.4.0-rc3)

erwango, 2023-10-12:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/57011

closed 2023-10-12 by erwango