zephyrproject-rtos/zephyr #59173: Bluetooth: Controller: No CIS disconnection event if CIS failed to be established on peripheral

Thalley, 2023-06-12:
**Describe the bug**
If the peripheral host accepts a CIS request that then fails to be established, the host seemingly never receives a Disconnect event. 

**To Reproduce**
N/A

**Expected behavior**
The spec states:

> When the Controller receives the HCI_LE_Accept_CIS_Request command, the Controller sends the HCI_Command_Status event to the Host. An HCI_LE_CIS_Established event will be generated when the CIS is established or is considered lost before being established.

**Impact**
Can put the host into a non-working state, which requires a complete reset. 

**Logs and console output**N/A

**Environment (please complete the following information):**
 - Commit SHA or Version used: 1b7ef97c31f4b734cae9cfb2b251c1b86b40d109

**Additional context**
N/A

cvinayak, 2024-01-24:
@Thalley Does this commit influence this GH issue? Was the host receiving the Failed to be Established event from the controller? https://github.com/zephyrproject-rtos/zephyr/commit/db588104921ac8bfd3f4c2b1b41c84cf5e710ec2

cvinayak, 2024-01-24:
@erbr-ot Could you check if Failed to be Established reason will be returned to Peripheral Host by this code:

https://github.com/zephyrproject-rtos/zephyr/blob/114548ef607aefd712dda7e5ea954981e778185b/subsys/bluetooth/controller/ll_sw/ull_llcp_cc.c#L254-L287

Controller is doing this:
https://github.com/zephyrproject-rtos/zephyr/blob/114548ef607aefd712dda7e5ea954981e778185b/subsys/bluetooth/controller/ll_sw/ull_llcp.c#L1347-L1366

Thalley, 2024-01-24:
> @Thalley Does this commit influence this GH issue? Was the host receiving the Failed to be Established event from the controller? [db58810](https://github.com/zephyrproject-rtos/zephyr/commit/db588104921ac8bfd3f4c2b1b41c84cf5e710ec2)

No, the commit from Rubin only affects the host, and this issue is for the controller

closed 2025-11-08 by jhedberg