zephyrproject-rtos/zephyr #59185: boards: lpcxpresso55s69: custom pinmux code in board file

gmarull, 2023-06-13:
**Describe the bug**

Board contains hardcoded pinmux related configuration. This should be handled somehow by the pinctrl driver.

**To Reproduce**

https://github.com/zephyrproject-rtos/zephyr/blob/bfa0a87277c31d9e1b36104a29b17f3fc1a47444/boards/arm/lpcxpresso55s69/pinmux.c#L12-L55

**Expected behavior**

Boards should not have pinmux related code.

**Impact**

N/A

**Logs and console output**

N/A

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.1
 - Commit SHA or Version used bfa0a87277c31d9e1b36104a29b17f3fc1a47444

**Additional context**
Discovered as part of a round of SYS_INIT usage cleanups

henrikbrixandersen, 2023-06-14:
Even though the function contains the word pinmux, this looks more like custom clock control?

decsny, 2024-02-12:
I think the function name here is wrong, because this is not actually setting up pinmux

decsny, 2024-02-12:
NXP already discussed internally how we want to move this code into DT somehow but have not had a chance to look into it yet. This file is not the only place where sysctl is configured by kconfig like this, I have made an issue to track this enhancement desire #68903

gmarull, 2024-02-13:
The code seems to be actually related to pinctrl, eg:

```
 #if (DT_NODE_HAS_COMPAT_STATUS(DT_NODELABEL(flexcomm6), nxp_lpc_i2s, okay)) && \ 
 		(DT_NODE_HAS_COMPAT_STATUS(DT_NODELABEL(flexcomm7), nxp_lpc_i2s, okay)) && \ 
 		CONFIG_I2S 
 	/* 
 	 * Flexcomm 6 and 7 are connected to codec on board, and shared signal 
 	 * sets are used to enable one I2S device to handle RX and one to handle 
 	 * TX 
 	 */ 
 	CLOCK_EnableClock(kCLOCK_Sysctl); 
 	/* Set shared signal set 0 SCK, WS from Transmit I2S - Flexcomm 7 */ 
 	SYSCTL->SHAREDCTRLSET[0] = SYSCTL_SHAREDCTRLSET_SHAREDSCKSEL(7) | 
 				SYSCTL_SHAREDCTRLSET_SHAREDWSSEL(7); 
```

Seems like you're setting sharing of SCK/WS signals. Can't this be handled by pinctrl driver itself? Looks like any board based on this SoC will need to have such hacks at board level otherwise.

decsny, 2024-02-13:
> The code seems to be actually related to pinctrl, eg:
>
> ```
>  #if (DT_NODE_HAS_COMPAT_STATUS(DT_NODELABEL(flexcomm6), nxp_lpc_i2s, okay)) && \
>  		(DT_NODE_HAS_COMPAT_STATUS(DT_NODELABEL(flexcomm7), [...]

The system control block is a different IP than the pinctrl IP. We have platforms that have this system ctrl IP but a different pinctrl IP ie different driver. Some driver for the system control block IP is a feature we don't have yet, we discussed it before internally and we just didn't have time to address this yet.

I completely agree I don't want this code in the board level as a hack and we talked about this before internally, which is why I just made the enhancement issue about it now that you also brought it up, but I do not think this can be solved by just adding something to the iocon pinctrl driver right now, as that would also be a hack

I prefer to keep the technical debt contained to the platform files rather than spread across the tree in different drivers

decsny, 2024-03-08:
The "pinmux" misnomer was fixed during HWMv2. This issue is not a bug, it's a feature request. Very simply, sysctl block is not a pinctrl IP and therefore cannot be handled by this pinctrl driver. The enhancement request to add sysctl support to DT is already being tracked by this issue https://github.com/zephyrproject-rtos/zephyr/issues/68903, so I am closing this one as a repeat.

PR #68898 (2024-03-12): boards: lpcxpresso55s69: Fix misnomer
pinmux.c is inaptly named since it has nothing to do with pinmux. Also, remove inclusion of iocon.h since this file does not use it, as this file has nothing to do with pinmux.

Fixes #59185
closed 2024-03-08 by decsny