zephyrproject-rtos/zephyr #59187: boards: mec2016evb_assy6797: uses custom pinmux scheme

gmarull, 2023-06-13:
**Describe the bug**

Board still uses a custom pinmux scheme, but this is no longer allowed and pinctrl API needs to be used.

**To Reproduce**

https://github.com/zephyrproject-rtos/zephyr/blob/bfa0a87277c31d9e1b36104a29b17f3fc1a47444/boards/arm/mec2016evb_assy6797/pinmux.c#L13-L38

**Expected behavior**

Pinctrl is used

**Impact**

N/A

**Logs and console output**

N/A

**Environment (please complete the following information):**

- OS: Linux
- Toolchain: Zephyr SDK 0.16.1
- Commit SHA or Version used https://github.com/zephyrproject-rtos/zephyr/commit/bfa0a87277c31d9e1b36104a29b17f3fc1a47444

**Additional context**

Discovered as part of a round of SYS_INIT usage cleanups

jvasanth1, 2023-06-15:
We need to remove mec2016 board - we are not supporting that

gmarull, 2023-06-16:
@jvasanth1 can you submit a PR deleting it?

jvasanth1, 2023-06-19:
> 

yes, we will do so

jvasanth1, 2023-06-19:
> @jvasanth1 can you submit a PR deleting it?

yes, we will do so

PR #59751 (2023-07-01): soc: MEC1701: Removed Microchip MEC1701
MEC1701 SOC specific sources has been removed.

Fix: #59187
closed 2023-07-01 by carlescufi