zephyrproject-rtos/zephyr #59219: Facing challenges in sysbuild flashing using west flash on nucleo_u575zi_q

PragatiGarg-eaton, 2023-06-14:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/59208

<div type='discussions-op-text'>

<sup>Originally posted by **PragatiGarg-eaton** June 13, 2023</sup>
Hi, 

I am working on building and flashing MCUBOOT+ application using sysbuild on nucleo_u575zi_q.

- nucleo_u575zi_q board is configured to be flashed using west STM32CubeProgrammer runner. 
- I am facing challenges in west flashing. After building the signed application along with MCUBOOT using sysbuild command, it is performing mass erase every time it flashes either mcuboot or signed image.
- And ideally, when invoking west flash on a build consisting of multiple images, each image is flashed in sequence without any chip erase.

west flash mcuboot:
[image]

west flash of signed image:
[image]
[image]

Commands used to build and flash:
build command: `west build -b nucleo_u575zi_q -p --sysbuild -- -DCONFIG_<mcuboot>=<y> --DCONFIG_<local signing>=<y>`
flash: `west flash`
Question: How can we disable the mass erase option with every west flash here?

OS: Windows
zephyr version: 3.2.0
STM32CubeProgrammer: 2.11.0

Links:
https://docs.zephyrproject.org/latest/build/sysbuild/index.html
https://docs.zephyrproject.org/3.0.0/boards/arm/nucleo_u575zi_q/doc/nucleou575zi_q.html#programming
</div>

nordicjm, 2023-06-14:
Message from discord on the issue from @erwango:
> Erase happens because of this line in board's board.cmak file: board_runner_args(stm32cubeprogrammer "--erase" "--port=swd" "--reset-mode=hw")
> Removing the --erase should solve your issue

erwango, 2023-06-14:
Closing as already discussed in #59208

closed 2023-06-14 by erwango