zephyrproject-rtos/zephyr #59221: doc: develop/application enhancement related to -DBOARD_ROUTE

rjmaris, 2023-06-14:
**Is your enhancement proposal related to a problem? Please describe.**
A path description for BOARD_ROUTE may be confusing for the user.

**Describe the solution you'd like**
See attached patch & https://github.com/zephyrproject-rtos/zephyr/discussions/58900

**Additional context**
-DDTC_OVERLAY_FILE
[0001-doc-add-hints-regarding-using-DBOARD_ROUTE.patch.txt](https://github.com/zephyrproject-rtos/zephyr/files/11743066/0001-doc-add-hints-regarding-using-DBOARD_ROUTE.patch.txt)

dbaluta, 2023-06-14:
Is this BOARD_ROUTE or BOARD_ROOT? Also, I think it's better to send the patch as a Pull Request.

rjmaris, 2023-06-14:
I already found the mistake before, but forgot to invoke git-format-patch again.
(regarding Pull Request: I don't mainain a fork)
[0001-doc-add-hints-regarding-using-DBOARD_ROOT.patch.txt](https://github.com/zephyrproject-rtos/zephyr/files/11743332/0001-doc-add-hints-regarding-using-DBOARD_ROOT.patch.txt)

rjmaris, 2023-06-14:
No problem. This is my contribution.
Some time ago this was principally discussed here: https://github.com/zephyrproject-rtos/zephyr/discussions/52533#discussioncomment-4274579
Quote from this: "Nobody prohibits sending patches as you did with https://github.com/zephyrproject-rtos/zephyr/issues/52556."

rjmaris, 2023-06-21:
Apparently, it's only me who's been a bit confusing about the path issue...

rjmaris, 2023-06-21:
From the patch everybody can extract/read exactly what is proposed to change. In the process of PR-processing a change of the proposed change could of course take place. Insofar no substantial change to proposiing it here. I admit that it would be better that I had started the issue simply with the **proposal** to change this docpage. From this issue I've learnt that I need to formulate my texts more appropriately.
Of course I'm not specifically interested in having my 'own' signed-off in a future patch. Anybody can do so.