zephyrproject-rtos/zephyr #59253: External flash mx25l2564 issue with stm32 Nucleo_h723zg

tejueaton, 2023-06-15:
Hi All,

I am interfacing the external flash mx25l2564 with nucleo_h723zg in Zephyr version 3.2 , Have connected the external flash Qspi in MCU,
Since in stm32h723vgtx-pinctrl.dtsi file those pins are mapped to octospi pins I have added this in my dts file ,

&octospi1{
    pinctrl-names = "default";
    pinctrl-0 = <&octospim_p1_clk_pb2 &octospim_p1_ncs_pb6
        &octospim_p1_io0_pc9 &octospim_p1_io1_pc10
        &octospim_p1_io2_pe2 &octospim_p1_io3_pd13>;
    status = "okay";

    mx25l2564: mx25l2564@0 {
        compatible = "st,stm32-ospi-nor";
        reg = <0>;
        ospi-max-frequency = <8000000>;
        size = <DT_SIZE_M(256)>; /* 32 MBytes */
		data-rate = <OSPI_STR_TRANSFER>;
        status = "okay";
        spi-bus-width = <4>;
        writeoc = "PP_1_4_4";
		sfdp-bfp = [
			53 46 44 50 06 01 02 ff
			00 06 01 10 30 00 00 ff
			c2 00 01 04 10 01 00 ff
			84 00 01 02 c0 00 00 ff
			00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00
			e5 20 fb ff ff ff ff 1f
			44 eb 08 6b 08 3b 04 bb
			fe ff ff ff ff ff 00 ff
			ff ff 44 eb 0c 20 0f 52
			10 d8 00 ff d6 49 c5 00
			81 df 04 e3 44 03 67 38
			30 b0 30 b0 f7 bd d5 5c
			4a 9e 29 ff f0 50 f9 85
			00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00
			00 00 00 00 00 00 00 00
			7f ef ff ff 21 5c dc ff
		];
		partitions {
			   compatible = "fixed-partitions";
			   #address-cells = <1>;
			   #size-cells = <1>;

			   partition@0 {
			       reg = <0x00000000 DT_SIZE_M(32)>;
			   };
		};
    };
};

and also have added these below lines in stm32h723.dtsi file.
```
clock-names = "ospix", "ospi-ker", "ospi-mgr";
            clocks = <&rcc STM32_CLOCK_BUS_AHB3 0x00004000>,
                <&rcc STM32_SRC_PLL1_Q OSPI_SEL(1)>,
                <&rcc STM32_CLOCK_BUS_AHB3 0x00200000>;
```

Getting error in running flash_shell sample writing and erasing to the flash.
Below is the output which i am getting :

**uart:~ *** Booting Zephyr OS build   ***
Found flash controller mx25l2564@0.
Flash I/O commands can be run.
[00:00:00.100,000] <inf> flash_stm32_ospi: OSPI flash config is SPI|DUAL|QUAD / STR
uart:~
uart:~$ flash erase mx25l2564@0 0 1000
Erase Failed, code -16.
[00:00:24.599,000] <err> flash_stm32_ospi: OSPI AutoPoll failed
[00:00:24.599,000] <err> flash_stm32_ospi: Erase failed : flash busy**

FRASTM, 2023-06-15:
Environment : nucleo_h723zg board with QSPI  mx25l2564 NOR flash  connected on the

erwango, 2023-06-15:
Converting to discussion until it is confirmed that issue is a genuine driver bug (potential dts config issue)