zephyrproject-rtos/zephyr #59282: Contributor nomination for GRobertZieba

GRobertZieba, 2023-06-15:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Robert Zieba
1. GitHub username: GRobertZieba
1. Organization (optional): Google

## Supporting Documents

Add links to 3-5 GitHub pull requests, in the Zephyr project, authored or
reviewed by the GitHub user that demonstrate the user's dedication to the
Zephyr project.

Reviewed:
https://github.com/zephyrproject-rtos/zephyr/pull/56872
https://github.com/zephyrproject-rtos/zephyr/pull/56959
https://github.com/zephyrproject-rtos/zephyr/pull/57568
https://github.com/zephyrproject-rtos/zephyr/pull/57956

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

keith-zephyr, 2023-07-06:
@carlescufi are you able to process this request?

closed 2023-07-12 by nashif

jfischer-no, 2023-07-12:
I am curious about this process, so is it okay to just approve 4 PRs to be added?

carlescufi, 2023-07-12:
> I am curious about this process, so is it okay to just approve 4 PRs to be added?

I was also under the impression that the contributor had to be an author of the PRs.

keith-zephyr, 2023-07-13:
The nomination template indicates 3-5 PRs authored or reviewed.

This nomination process only elevates the user from the Read to Triage permission level.  I don't think we need a very high bar here as it's hard to engage with users that only have Read permission on the project.  Read-only users cannot add reviewers to their PRs, and you cannot add the Read-only users as a reviewer to other PRs.