zephyrproject-rtos/zephyr #59289: zcbor_map_decode_bulk return code is unlike the other API functions

WayneKeenan, 2023-06-15:
Shouldn't the enclosing function just return the bool from `zcbor_map_end_decode`?

https://github.com/zephyrproject-rtos/zephyr/blob/68dc53b0779a37342f86a29362c8efdafdcea005/subsys/mgmt/mcumgr/util/src/zcbor_bulk.c#L71

WayneKeenan, 2023-06-15:
or not be prefixed `zcbor` as it's from `mcumgr` and behaves more like the Zephyr API.
or be migrated into core `zcbor_decode` and apply the same semantics as `zcbor`

nordicjm, 2023-06-16:
This function is for internal MCUmgr use, as the path suggests, it is not part of the zcbor library/code.

WayneKeenan, 2023-06-16:
It is it in the migration guide. 

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/migration/migration_guide_1.x_to_2.x.html

nordicjm, 2023-06-16:
Indeed that text is wrong, it should be `int rc;` and `rc = ...` not `ok = ...` @peknis

WayneKeenan, 2023-06-16:
One could argue, in C++ parlance,  it's more of a 'protected' function than 'private'.    
In which case it would be nice if it behaved in the same way as the 'public' functions.

WayneKeenan, 2023-06-16:
"Path of least surprise"

WayneKeenan, 2023-06-16:
It's designed for custom implementations of MCUmgr too, not just the provided one.

Ok, happy to close.

closed 2023-06-16 by WayneKeenan

nordicjm, 2023-06-16:
Actually the documentation is correct:
```
ok = zcbor_map_decode_bulk(zsd, image_upload_decode,
                           ARRAY_SIZE(image_upload_decode), &decoded) == 0;
```
If `zcbor_map_decode_bulk` returns 0 (success) then `ok` will be set true, otherwise it will be set false