zephyrproject-rtos/zephyr #59303: Add json-c as external module

parthitce, 2023-06-16:
## Origin

JSON-C implements a reference counting object model that allows you to easily construct JSON objects in C, output them as JSON formatted strings and parse JSON formatted strings back into the C representation of JSON objects. It aims to conform to [RFC 7159](https://tools.ietf.org/html/rfc7159).

URL: https://github.com/json-c/json-c

## Purpose

json-c provides friendly API to frame and parse json messages. It's well known in the Linux world and bringing it helps,

- building cross platform (compatible between Linux and zephyr apps)
- Friendly dynamic processing of complex json objects

Comparing with available native json data handling, json-c helps,
- framing json payloads using heap memory
- No need to statically pre-built json structures

## Mode of integration

Mode of integration: External module
Repository name: json-c
Reason: To keep in sync with the open source version of json-c, which helps rebasing

## Maintainership

- json-c open source community
- @ssekar15 
- @parthitce 

## Pull Request

Yet to prepare a final one based on https://github.com/linumiz/json-c/tree/zephyr

## Description

Please refer to https://json-c.github.io/json-c/

What is its primary functionality?
- JSON parsing and framing with friendly API's with extended functionality

What problem are you trying to solve?
- JSON processing with existing native implementation needs static definition of the json structures
- Dynamic parsing and framing of json objects

Why is this the right component to solve it?
- Predominant in use in Linux world and well known for it's stability
- ~18K footprint in memory (text + data + bss)

## Dependencies

What other components does this package depend on?
- None

Will the Zephyr project have a direct dependency on the component, or
will it be included via an abstraction layer with this component as a
replaceable implementation?
- We will have abstraction for file handling if needed based on LFS

## Revision

Version or SHA you would like to integrate initially
- https://github.com/json-c/json-c/releases/tag/json-c-0.16-20220414

## License

Please use an SPDX identifier (https://spdx.org/licenses/), such as
``BSD-3-Clause``
- https://github.com/json-c/json-c/blob/master/COPYING
- MIT License

nashif, 2023-06-21:
@mrfuchs  can you please take a look?

parthitce, 2023-06-28:
@mrfuchs Ping! Thanks.

parthitce, 2023-07-05:
Added the initial draft PR for review https://github.com/zephyrproject-rtos/zephyr/pull/60039

parthitce, 2023-07-24:
@nashif @mrfuchs Ping for json-c!

parthitce, 2023-08-28:
@nashif @mrfuchs Gentle ping! Please share your feedback and comments.

parthitce, 2023-12-21:
@nashif @mrfuchs Please have a look. Thanks.

keith-zephyr, 2024-04-16:
This issue looks like it fell through the cracks and wasn't looked at by the TSC.

parthitce, 2024-04-17:
> This issue looks like it fell through the cracks and wasn't looked at by the TSC.

Thanks, let me know if further details are required to discuss with TSC.
Active PR: https://github.com/zephyrproject-rtos/zephyr/pull/66763

carlescufi, 2024-05-08:
> Based on the in-person discussion by @ssekar15 with you Carles @carlescufi at EOSS, could you please help us proceed with this issue? We would like to get this json-c implementation get into Zephyr. Also please let us know the process and to join the TSC to discuss about this issue. Many thanks.

The process:
https://docs.zephyrproject.org/latest/contribute/external.html

Joining the TSC:
https://github.com/zephyrproject-rtos/zephyr/wiki/Technical-Steering-Committee-%28TSC%29

But I have to repeat what I said in person: we cannot have two implementations of the same functionality (JSON) in Zephyr. This means that, for this to be included in Zephyr, we have to deprecate and remove the current JSON library in the main Zephyr tree. To do that we need to make sure that we select the right replacement for it. So we would need a bit more info regarding json-c and how it compares to both the existing Zephyr library and other external projects that also implement the same, like for example https://github.com/DaveGamble/cJSON.

parthitce, 2024-08-06:
> The process: https://docs.zephyrproject.org/latest/contribute/external.html
>
> Joining the TSC: https://github.com/zephyrproject-rtos/zephyr/wiki/Technical-Steering-Committee-%28TSC%29
>
> But I have to [...]

@carlescufi thanks for reviewing this. With external optional modules support, does moving it as optional module under submanifest makes sense? This way it's not pulled in by default and used only when when the user/developer enables the optional?

I understand the primary question about the comparison with best replacement. As far as json-c is concerned, it uses heap. I didn't profile yet the usage of heap, but do we have a way to profile heap in zephyr? 
One other question is, modules of this kind can use heap?

parthitce, 2024-09-07:
After careful evaluation of memory footprint with json-c, am closing this issue. The main reasons are,

- Very recursive small bin usage with malloc in json-c
- Possible replacement with cjson is already handled in nrf-sdk, which might be better alternative (but not evaluated)

To keep the existing work for json-c, for future reference and also potential use for devices with more RAM (SRAM/PSRAM), implementation is moved to https://github.com/linumiz/rixx

closed 2024-09-07 by parthitce