zephyrproject-rtos/zephyr #59317: Assertion fail after aborting Bluetooth Mesh provisioning

mrfuchs, 2023-06-16:
Aborting provisioning results in an [assertion fail](https://github.com/zephyrproject-rtos/zephyr/blob/6981e2e3cf636119a41693696dbe5a184be808af/subsys/bluetooth/host/hci_core.c#L330) when running Bluetooth Mesh on the nrf5340dk_nrf5340 board.

Board: nrf5340dk_nrf5340
Sample projects:
 * cpuapp: [samples/bluetooth/mesh](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/mesh)
 * cpunet: [samples/bluetooth/hci_rpmsg](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/hci_rpmsg)

Version: Zephyr 3.4.0-rc3 (d14519c)

To reproduce the issue, abort the provisioning like this: → Open "nRF Mesh" mobile app → Browse to "Network" tab -> Tap "+" on upper right corner → Select "Zephyr" device → Tap "Abort" on upper left corner → Tap "Cancel" on upper left corner

cpuapp will crash with this debug output:

```
[00:00:00.400,726] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
*** Booting Zephyr OS build d14519cb24a1 ***
Initializing...
Set up button at gpio@842500 pin 23
[00:00:00.420,501] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.420,501] <inf> bt_hci_core: HW Variant: nRF53x (0x0003)
[00:00:00.420,532] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.4 Build 0
[00:00:00.422,149] <inf> bt_hci_core: Identity: C1:ED:90:D7:B9:FE (random)
[00:00:00.422,180] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:00.422,210] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
Bluetooth initialized
Mesh initialized
[00:00:21.730,010] <wrn> bt_mesh_pb_gatt_srv: Client wrote 0x0000 instead enabling notify
[00:00:21.730,804] <wrn> bt_hci_core: OpCode 0x0c35 completed instead of expected 0x2039
[00:00:21.731,018] <wrn> bt_hci_core: opcode 0x2039 pool id 6 pool 0x20001354 != &hci_cmd_pool 0x200013ac
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:330
  command opcode 0x2039 timeout with err -11
[00:00:31.730,834] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000002
[00:00:31.730,865] <err> os: r3/a4:  0x20001878 r12/ip:  0x00000010 r14/lr:  0x00004e69
[00:00:31.730,865] <err> os:  xpsr:  0x41000000
[00:00:31.730,895] <err> os: Faulting instruction address (r15/pc): 0x00004e74
[00:00:31.730,926] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:31.730,957] <err> os: Current thread: 0x20003698 (sysworkq)
[00:00:32.007,537] <err> os: Halting system
```

As a workaround, I enabled the following Kconfig options (for samples/bluetooth/mesh):

```
CONFIG_BT_HCI_ACL_FLOW_CONTROL=n
CONFIG_BT_ATT_ENFORCE_FLOW=n
```

LingaoM, 2023-06-20:
[image]

LingaoM, 2023-06-20:
The host number of completed packets command assume not response by controller.

PavelVPV, 2023-06-26:
I wasn't able to reproduce the issue on commit `942ef53d08`. I compiled `hci_rpmsg` with `nrf5340_cpunet_bt_mesh-bt_ll_sw_split.conf` config, then `samples/bluetooth/mesh/` app with `nrf5340dk_nrf5340_cpuapp_ns` board, then provisioned using nRF Mesh app for Android and was able to send OnOff messages to server model from the mobile app, the board also reacted on button 1 presses.

mrfuchs, 2023-06-26:
Hi @PavelVPV,

Thanks for your suggestions.

> [...] could you please tell what board did you use for application core: `_cpuapp_ns` or `_cpuapp`?

I'm using `_cpuapp` and `_cpunet`, no `_ns`.

> Can you check if the extended advertiser is enabled on the network core (`CONFIG_BT_EXT_ADV` in `hci_rpmsg`)?

I was using `CONFIG_BT_CTLR_ADV_EXT=y`. Added `CONFIG_BT_EXT_ADV` to the `hci_rpmsg`'s configuration file. Still experiencing the assertion fail...

> I compiled hci_rpmsg with nrf5340_cpunet_bt_mesh-bt_ll_sw_split.conf config [...]

Tested it with this config on my side. I had to comment `CONFIG_BT_PHY_UPDATE=n` otherwise the nRF Mesh iOS app hangs at "Connecting..." (is this a regression of #14658?).
Once I commented (`#CONFIG_BT_PHY_UPDATE=n`) the project runs as good as before, i.e. I'm able to provision the device, but it crashes after aborting the provisioning.

> then provisioned using nRF Mesh app for Android and was able to send OnOff messages to server model from the mobile app, the board also reacted on button 1 presses

This also works for me. Did you try to abort provisioning using the (iOS) nRF Mesh app? That's when I see the kernel oops due to the assertion failure.

> I wasn't able to reproduce the issue on commit 942ef53d08.

Will try to update and retest.

PavelVPV, 2023-06-27:
We were not able to reproduce this issue neither on https://github.com/zephyrproject-rtos/zephyr/commit/356c8cbe63ae01b3ab438382639d25bb418a0213 commit nor on https://github.com/zephyrproject-rtos/zephyr/commit/a7498c39e2d34d33a463fc861f1f39f0a626b8da commit on iPhone 7 Plus with iOS 15.6.1.

What is iPhone and iOS version do you have?
Did you do other changes to the sample?
Could you try other bluetooth samples (not mesh samples)?

PavelVPV, 2023-06-28:
I also tried on iPad 8 gen with iOS 15.5 and couldn't reproduce the issue on 3.4.0 release.

kruithofa, 2023-07-06:
@mrfuchs We haven't been able to reproduce. Can you answer Pavel's questions

What is iPhone and iOS version do you have?
Did you do other changes to the sample?
Could you try other bluetooth samples (not mesh samples)?

mrfuchs, 2023-07-13:
@PavelVPV, @kruithofa, I decided to live with the workaround I mentioned above.

> What is iPhone and iOS version do you have?

iPhone 14 Pro running iOS 16.5.1.

> Did you do other changes to the sample?

Not that I am aware of.

> Could you try other bluetooth samples (not mesh samples)?

Well, other Bluetooth samples work for fine (like periodic_adv etc.), but that was to be expected, because I cannot use the nRF Mesh mobile app and abort provisioning with these samples - which is the only thing that does not work for me.

PavelVPV, 2023-07-19:
In the log you attached I see these messages:
```
[00:00:00.400,726] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
```
and
```
Set up button at gpio@842500 pin 23
```

The first one is printed if IEEE802154 enabled, but it is not enabled in the sample. Could you try the sample with pure prj.conf? Could you also provide the complete prj.conf and the commands you use to compile both the sample and hci_rpmsg?

The second one I can't find in the mesh sample.

golemik98, 2024-03-04:
Hi!
This issue still exists on Zephyr 3.4.0. Is there a fix on later versions? 
To reproduce this version we use iphone 12 and newer.
Would be great to receive and update on that, thanks! @PavelVPV @kruithofa @

PavelVPV, 2024-03-04:
> Hi! This issue still exists on Zephyr 3.4.0. Is there a fix on later versions? To reproduce this version we use iphone 12 and newer. Would be great to receive and update on that, thanks! @PavelVPV @kruithofa @

Hi @golemik98 , from what I recall I couldn't reproduce it with any of our samples. Could you please reproduce it with unmodified mesh sample? Could you try a newer Zephyr version? Could you provide more information on your setup?

golemik98, 2024-03-04:
Thanks for answer! 
I will try to reproduce it on 3.4.0 sample, I will let you know as soon as I will achieve it.

PavelVPV, 2024-03-07:
@golemik98, I've tried again with zephyr v3.4.0 and v3.6.0 both, secure and non-secure versions (except non-secure version on 3.4.0 as I wasn't able to flash all images for unknown reasons) but on iPad 8th gen as I used to test in the comment above. And this time on iOS 16.smth (probably 16.7) and 17.4. Always was able to provision the DK.

Could you please enable the following Kconfig options to see if there are any errors when provisioning a device:
```
diff --git a/samples/bluetooth/hci_ipc/prj.conf b/samples/bluetooth/hci_ipc/prj.conf
index 755a1f4ac1e..1bc9145c8d1 100644
--- a/samples/bluetooth/hci_ipc/prj.conf
+++ b/samples/bluetooth/hci_ipc/prj.conf
@@ -22,3 +22,6 @@ CONFIG_BT_BUF_CMD_TX_COUNT=10
 # CONFIG_BT_BUF_ACL_RX_SIZE=255
 # CONFIG_BT_BUF_ACL_TX_SIZE=251
 # CONFIG_BT_BUF_CMD_TX_SIZE=255
+
+CONFIG_LOG=y
+CONFIG_BT_LOG_LEVEL_WRN=y
diff --git a/samples/bluetooth/mesh/prj.conf b/samples/bluetooth/mesh/prj.conf
index 9c8daad9131..3a26a43ccfd 100644
--- a/samples/bluetooth/mesh/prj.conf
+++ b/samples/bluetooth/mesh/prj.conf
@@ -34,3 +34,6 @@ CONFIG_BT_MESH_SUBNET_COUNT=2
 CONFIG_BT_MESH_APP_KEY_COUNT=2
 CONFIG_BT_MESH_MODEL_GROUP_COUNT=2
 CONFIG_BT_MESH_LABEL_COUNT=3
+
+CONFIG_LOG=y
+CONFIG_BT_HCI_DRIVER_LOG_LEVEL_WRN=y
```

carlescufi, 2024-04-11:
Closing due to inactivity from reporter. @mrfuchs please reopen if you can provide the required logs.

closed 2024-04-11 by carlescufi