zephyrproject-rtos/zephyr #59322: Newlib memcpy happens to do unaligned memory access

yeaissa, 2023-06-16:
**Is your enhancement proposal related to a problem? Please describe.**
NewLib c memcpy function happens to do unaligned memory access. This may produce faults when accessing a memory region that doesn’t support this access.

**Describe the solution you'd like**
Maybe the memcpy function should do byte-sized copying when unaligned memory access is requested.

stefanct, 2024-03-15:
You might be interested in https://github.com/nxp-mcuxpresso/mcux-sdk/blob/675a70e9b9ea5de2177f8881c31f464e0cb30528/utilities/misc_utilities/fsl_memcpy.S
It has some more information and a working implementation under BSD-3-Clause (I can't advocate for it, I just noticed it).

gdb3-ampaworks, 2024-08-01:
I am running into this issue as well -- trying to use external device memory (OSPI SRAM) for network buffer space. Periodically, an unaligned call to `memcpy` will occur and cause a bus fault.

Shouldn't we have a `memcpy` implementation that handles this? More and more, embedded systems are utilizing external memory devices that don't support unaligned access...