zephyrproject-rtos/zephyr #59340: Encounter build error caused by redefinition of declarations on Windows

MulinChao, 2023-06-19:
**Describe the bug**
A build error caused by redefinitions of `c macro` and `function declarations`.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples\hello_world
2. set BOARD=npcx9m6f_evb
3. west build -p always
4. See error

**Expected behavior**
build successfully.

**Logs and console output**
Please refer attached [build_error.zip](https://github.com/zephyrproject-rtos/zephyr/files/11784865/build_error.zip). We found lots of  redefined warnings/errors in `syscall_list.h` and `kernel.h` generated by `gen_syscalls.py`
```
In file included from C:/Users/TomChang/Desktop/zephyrproject/zephyr/include/zephyr/syscall.h:11,
                 from C:/Users/TomChang/Desktop/zephyrproject/zephyr/include/zephyr/kernel_includes.h:31,
                 from C:/Users/TomChang/Desktop/zephyrproject/zephyr/include/zephyr/kernel.h:17,
                 from C:/Users/TomChang/Desktop/zephyrproject/zephyr/arch/arm/core/offsets/offsets_aarch32.c:28,
                 from C:/Users/TomChang/Desktop/zephyrproject/zephyr/arch/arm/core/offsets/offsets.c:9:
C:/Users/TomChang/Desktop/zephyrproject/zephyr/samples/hello_world/build/zephyr/include/generated/syscall_list.h:215: warning: "K_SYSCALL_CLOCK_GETTIME" redefined
  215 | #define K_SYSCALL_CLOCK_GETTIME 206
      | 
C:/Users/TomChang/Desktop/zephyrproject/zephyr/samples/hello_world/build/zephyr/include/generated/syscall_list.h:6: note: this is the location of the previous definition
    6 | #define K_SYSCALL_CLOCK_GETTIME 0
```

**Environment (please complete the following information):**
 - OS:  Windows
 - Toolchain - Zephyr SDK, 15.1
 - Commit SHA: 32fae07f299aa326

nordicjm, 2023-06-19:
@dcpleung seems possibly related to your recent change?

ndias88, 2023-06-20:
Issue fixed on my side after applying commit [59350](https://github.com/zephyrproject-rtos/zephyr/pull/59350/commits)
Time to rock on zephyr 3.4. Thanks :)

PR #59350 (2023-06-19): scripts: build: parse_syscalls: Fix duplicate paths on windows
    Fixes an issue on windows where there would be many permutations
    of paths with different folder separators by converting to
    posix-style path separators.
    
    Issue was introduced by
    https://github.com/zephyrproject-rtos/zephyr/pull/58351

Fixes #59340

**Needs verification to ensure this is the proper fix.**