zephyrproject-rtos/zephyr #59355: Flash Erase on STM32L4

JulienGodt, 2023-06-19:
**Describe the bug**
I created two slots for each of my images and a partition for my bootloader like this in my DTS:
`
&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "bootloader";
			reg = <0x000000000 0x00020000>;
		};
		keystore_partition: partition@20000 {
			label = "keystore";
			reg = <0x00020000 0x00001000>;
		};
		params_partition: partition@21000 {
			label = "params-storage";
			reg = <0x00021000 0x0000f000>;
		};
		slot0_partition: partition@30000 {
			label = "image-0";
			reg = <0x00030000 0x00060000>;
		};
		slot1_partition: partition@90000 {
			label = "image-1";
			reg = <0x00090000 0x00060000>;
		};
		storage_partition: partition@f0000 {
			label = "storage";
			reg = <0x000f0000 0x00010000>;
		};
	};
	reg = <0x08000000 DT_SIZE_K(1024)>;
};`

When I try to erase my flash, I can't do it even though I tried several ways.
Here is my test code:
`
static void flash_test(void)
{
    uint8_t data[64];
    uint8_t buf[64];
    int ret;
    size_t size = 64;
    struct flash_pages_info info;

    off_t offset = 0x30000;

    static const struct device *flash_device;
    flash_device =
      DEVICE_DT_GET(DT_CHOSEN(zephyr_flash_controller));
    if (flash_device) {
      printk("Found flash controller.\n");
      printk("Flash I/O commands can be run.\n");
    } else {
      printk("**No flash controller found!**\n");
      printk("Run set_device <name> to specify one "
            "before using other commands.\n");
    }

    ret = flash_get_page_info_by_offs(flash_device, offset, &info);
    if (ret) {
      return ret;
    }
    LOG_DBG("info.start_offset 0x%x, info.size %d, info.index %d", info.start_offset, info.size, info.index);

    printf("Flash offset: %.X \r\n", offset);

    for(int i = 0; i < 64; i++) {data[i]= i;}

    ret = flash_erase(flash_device, offset, size);
    if (ret) {
      printk("flash_erase failed (err:%d).\n", ret);
      return;
    }

    ret = flash_write(flash_device, offset, &data, sizeof(data));
    if (ret) {
      printk("flash_write failed (err:%d).\n", ret);
      return;
    }

    ret = flash_read(flash_device, offset, buf, sizeof(buf));
    if (ret) {
        printf("Error reading from flash \r\n");
    }
    else {
        printf("Reading from flash .. \r\n");
        for (int k = 0; k < sizeof(buf); k++) {
            printf("%.2x ", buf[k]);
        }
        printf("\r\n");
    }

    printk("Success");

}`

**To Reproduce**
Steps to reproduce the behavior:
1. west build --pristine -b stm32l4q5@B
2. west flash

**Expected behavior**
I'm trying to write to the slot I'm not using. I am currently running on slot B, and would like to write to slot A.

**Logs and console output**
\`\`\`
00> Found flash controller.
00> Flash I/O commands can be run.
00> [00:00:57.244,000] <dbg> boot_image: flash_test: info.start_offset 0x30000, info.size 4096, info.index 48
00> [00:00:57.266,000] <err> flash_stm32l4: Word at offs 196608 not erased
00> flash_write failed (err:-5).
\`\`\`

**Environment :**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Zephyr Version 3.4.0-rc3
 - Target: STM32L4Q5CG 

**Additional context**
I start with zephyr, be indulgent...

Laczen, 2023-06-19:
@JulienGodt, you cannot erase the flash with a size of 64, the size supplied to `flash_erase` needs to be a multiple of the erase-block size.

JulienGodt, 2023-06-20:
Thank you for your answer,
Several things were wrong, the size was not aligned with the erase block-size, plus I couldn't erase the entire slot0, straddling bank 1 and 2.
One last quick question:
When I erase even 2 pages, the stm32 does not boot anymore. My bootloader being on bank 0, do you know if there is a flag to modify or a register?

closed 2023-07-11 by jhedberg