zephyrproject-rtos/zephyr #59364: Bluetooth: Host: Fail to initiate a connection to a synced paired device from a PAwR train if BT_PRIVACY=y

laxiLang, 2023-06-19:
**Describe the bug**
Do pairing before initiating a connection to a synced paired device from a PAwR train will lead to the connection v2 failing to be established. The reason is that on the synced device, the device address is the resolvable private address, while the local identity address is used as the peer address by the controller when establishing a connection from PAwR. 

**To Reproduce**
Steps to reproduce the behavior:
1. both devices create new identity with device address and IRK pairs
2. both devices call bt_enable()
3. connect the two devices and set security level 2 for adding peer device to resolving list
4. on device starting PAwR adverting while the other syncs to it
5. the PAwR advertiser initiates a connection to the synced device

**Expected behavior**
The connection v2 initiation will get 0x3E (Failed to be established) error

**Impact**
This will affect the establishment of PAwR connection v2 with privacy enabled

jori-nordic, 2023-06-21:
@laxiLang moved to low as this is still experimental in Zephyr.

alwa-nordic, 2023-06-21:
@hermabe FYI if you are still working on PAwR :)

alwa-nordic, 2023-06-21:
Is this related to PR https://github.com/zephyrproject-rtos/zephyr/pull/57668?

hermabe, 2023-06-21:
> Is this related to PR #57668?

That is a separate issue.

I am aware of this but have not looked too much into it.

closed 2024-01-06 by github-actions[bot]