zephyrproject-rtos/zephyr #59383: `GPIO_OUTPUT_INACTIVE` and `GPIO_OUTPUT_LOW` do not work as expected.

zfields, 2023-06-19:
**Describe the bug**
`GPIO_OUTPUT_INACTIVE` does not set the associated GPIO pin to the inactive value.
`GPIO_OUTPUT_LOW` does not set the associated GPIO pin to the low value.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
     swan_r5
 - What have you tried to diagnose or workaround this issue?
     `gpio_pin_toggle_dt` works as expected
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
     I have not checked this.
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. modify the blink sketch in the manner shown below
2. See error

**Expected behavior**
I would expect a blinking LED (i.e. the exact same behavior as the unmodified `blinky` application).

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
I can work around with `gpio_pin_toggle_dt`, but this seems like a major problem with the platform.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
     Linux (Ubuntu 22.04)
 - Toolchain (e.g Zephyr SDK, ...)
     Zephyr SDK
 - Commit SHA or Version used
    - Zephyr version: 3.4.99
    - build: v3.4.0-2-g79d610a89338

**Additional context**
```c
/*
 * Copyright (c) 2016 Intel Corporation
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr/kernel.h>
#include <zephyr/drivers/gpio.h>

/* 1000 msec = 1 sec */
#define SLEEP_TIME_MS   1000

/* The devicetree node identifier for the "led0" alias. */
#define LED0_NODE DT_ALIAS(led0)

/*
 * A build error on this line means your board is unsupported.
 * See the sample documentation for information on how to fix this.
 */
static const struct gpio_dt_spec led = GPIO_DT_SPEC_GET(LED0_NODE, gpios);

int main(void)
{
	int ret;

	if (!gpio_is_ready_dt(&led)) {
		return 0;
	}

	ret = gpio_pin_configure_dt(&led, GPIO_OUTPUT_ACTIVE);
	if (ret < 0) {
		return 0;
	}

	while (1) {
		ret = gpio_pin_set_dt(&led, GPIO_OUTPUT_ACTIVE);
		if (ret < 0) {
			return 0;
		}
		k_msleep(SLEEP_TIME_MS);
		ret = gpio_pin_set_dt(&led, GPIO_OUTPUT_INACTIVE);
		if (ret < 0) {
			return 0;
		}
		k_msleep(SLEEP_TIME_MS);
	}
	return 0;
}
```

henrikbrixandersen, 2023-06-20:
Reassigning to @erwango since the `swan_r5` board uses a `stm32l4r5Xi` MCU.

zfields, 2023-08-20:
This is a rather serious issue. I would like for the `Stale` tag to be removed.

erwango, 2023-08-21:
Handing over ownership to @henrikbrixandersen as GPIO maintainer

zfields, 2023-10-23:
Please remove this label. This is incredibly fundamental and important functionality.

jfischer-no, 2023-10-24:
Macros GPIO_OUTPUT_*ACTIVE and GPIO_OUTPUT_LOW are to configure GPIO pin not to set it logical level. To set the pin to active `gpio_pin_set_dt()` expects a non-zero value and to set it to inactive expects a zero value.

https://github.com/zephyrproject-rtos/zephyr/blob/b510073db3c4537ba0dce882b8a522692cf00cd9/include/zephyr/drivers/gpio.h#L1594-L1599

closed 2023-10-25 by henrikbrixandersen