zephyrproject-rtos/zephyr #59405: Bluetooth API does not support C++ projects

jcondon-idl, 2023-06-19:
Currently, it appears that many of the macros utilized (and required) by the Bluetooth subsystem are not supported in C++. Specifically any of the macros that return a pointer to newly allocated memory fail. The object created only lives until the end of the expression so the returning pointer is to a temporary array e.g. (BT_UUID_DECLARE_128, BT_LE_ADV_CONN).

hermabe, 2023-06-20:
Because C++ has different lifetime requirements, you will have to create variables for these in your application. Both of your examples of macros are not required, but rather convenience macros.

```c++
// Use &uuid instead of BT_UUID_DECLARE_128(value)
struct bt_uuid_128 uuid = BT_UUID_INIT_128(value);

// Use &param instead of BT_LE_ADV_PARAM(_options, _int_min, _int_max, _peer)
struct bt_le_adv_param param = BT_LE_ADV_PARAM_INIT(_options, _int_min, _int_max, _peer);

// For BT_LE_ADV_CONN, specify the parameters like if you wanted something different than the default.
struct bt_le_adv_param connectable = BT_LE_ADV_PARAM_INIT(BT_LE_ADV_OPT_CONNECTABLE, BT_GAP_ADV_FAST_INT_MIN_2, BT_GAP_ADV_FAST_INT_MAX_2, NULL);
```

Is there any Bluetooth feature you are unable to use because of the use of C++'s different lifetime rules? Otherwise, I propose to close this as not a bug.

jhedberg, 2023-06-21:
As @hermabe points out, the macros you're trying to use are for convenience but only supported in C code (not C++). As these are not mandatory, you can simply skip using them in your C++ code.

closed 2023-06-21 by jhedberg

hermabe, 2023-07-17:
`BT_GATT_SERVICE_DEFINE` doesnt do anything illegal related to lifetimes. It creates an array of attributes directly from the inputs and creates an iterable section. The `bt_gatt_service` struct is just a list of attributes as defined in the Bluetooth Core spec. The undocumented `node` field is for stack-internal use.

`BT_GATT_ATTRIBUTE` does not do anything special, can be replaced by a designated initializer.

`BT_GATT_CHARACTERISTIC` does create a temprorary `bt_gatt_chrc`, so you will have to create a separate variable for that one in c++.

If you want to see what the macros do you can run for example `samples/bluetooth/peripheral` through the preprocessor and look at the output.

vChavezB, 2024-04-13:
I made a  POC to translate the GATT attribute structure initialization for an OOP approach.

https://github.com/vChavezB/ble_utils