zephyrproject-rtos/zephyr #59412: Json Floating Point Encoding

ofirshe, 2023-06-20:
Hey, 

To facilitate the encoding of float numbers, developers need to store fields of type 'struct json_obj_token' in their structures instead of directly using float variables. However, this approach can be inconvenient, especially when dealing with multiple float variables within the struct. In such cases, developers are required to maintain temporary auxiliary char buffers for each float variable, converting the float values to strings. Additionally, if there is a requirement to process these values, they can be utilized directly as float numbers without any conversion from 'struct json_obj_token' to float. Consequently, it is advisable to modify the function name from "float_ascii_encode" to "float_encode," which invokes the "snprintk" function with the format "%f" and apparently another changes required.

naNEQ, 2024-03-14:
Was anything done on this? I believe this is something that is indeed missing.

ofirshe, 2024-03-14:
Hey, 

you can apply the patch I've created—it hasn't been upstreamed.
[json_float_support.patch](https://github.com/zephyrproject-rtos/zephyr/files/14599205/json_float_support.patch)

naNEQ, 2024-03-15:
Thanks, I was also thinking along the same lines.

EricNRS, 2025-02-03:
Would love to see this fix merged and also fixes on the encode side.  Also, should changes be made to add support for doubles as well?

ofirshe, 2025-02-03:
Hi @EricNRS, this requires a significant change, and I'm currently swamped with work. I've implemented a temporary local fix for now. If you have the time to take care of it, that would be great. I believe many people are eagerly waiting for this enhancement.

EricNRS, 2025-02-03:
@ofirshe I'm swamped as well.  I pulled in your temporary fix (much appreciated) to kick the can down the road a bit further.

EricNRS, 2025-03-08:
Needed decode support, so pushed through getting support for float and double into the library.  Will push the fix in a few minutes.

EricNRS, 2025-03-08:
Also did a fix to allow using the JSON library from C++ for those brave souls:  https://github.com/zephyrproject-rtos/zephyr/pull/86692

EricNRS, 2025-03-12:
I have closed https://github.com/zephyrproject-rtos/zephyr/pull/86800.  Please use https://github.com/zephyrproject-rtos/zephyr/pull/86956 as the fix for this issue.  It is ready for final review.

commit fc37c02e (2025-04-09): json: improve parsing and serializing of 'float' and 'double'

Up to now, the handling of type float was offloaded to the users of the
JSON utility, with the token JSON_TOK_FLOAT.

Improve handling of floating point types and support the types 'float'
and 'double' in a built-in way so that they can be directly parsed to
and serialized from variables (of type float or double).

The types are serialized in the shortest representation, either as a
decimal number or in scientific notation:
  * float (with JSON_TOK_FLOAT_FP): encoded with maximal 9 digits
  * double (with JSON_TOK_DOUBLE_FP): encod [... cut]
closed 2025-04-09 by kartben