zephyrproject-rtos/zephyr #59443: os: lib: spsc_pbuf: spsc_pbuf_free does not care if correct len is passed.

emob-nordic, 2023-06-20:
**Is your enhancement proposal related to a problem? Please describe.**
I think the API allows to exploit and break the functionality. `spsc_pbuf_free` function assumes there was a call to `spsc_pbuf_claim` before and that the user frees exactly the amount `spsc_pbuf_claim` claimed, there is no check on that.

With current API it is possible to free less space that was claimed.
Assume the following:
```
char *rx;
char rx_data[50];

uint16_t len = spsc_pbuf_claim(pb, &rx);
/* Assume there was someting in the buffer and len == 10, data points to valid memory. */
spsc_pbuf_free(pb, 5);

int len = spsc_pbuf_read(pb, rx_data, 50);
/* As per spsc_pbuf structure the returned len is some value from previous packet,
because we freed only 5 bytes before and moved rx index by not correct value. */

```

**Describe the solution you'd like**
I think `spsc_pbuf_free` should look like follows:

`void spsc_pbuf_free(struct spsc_pbuf *pb, char *buf, uint16_t len);`

**Describe alternatives you've considered**
Careful usage.

nordic-krch, 2023-06-20:
In you proposal what would be the reason for `buf` and `len` in the `spsc_pbuf_free` argument list? If it is to confirm if `len` is correct then it is redundant for me.
 
What about just `spsc_pbuf_free(struct spsc_pbuf *pb, char *buf)` or even `spsc_pbuf_free(struct spsc_pbuf *pb);` for freeing last packet (claimed or not).

emob-nordic, 2023-06-20:
> In you proposal what would be the reason for `buf` and `len` in the `spsc_pbuf_free` argument list? If it is to confirm if `len` is correct then it is redundant for me.
>
> What about just [...]

You are right. I didn't think about it this way. Imo `spsc_pbuf_free(struct spsc_pbuf *pb, char *buf)` is better option.
As a user i expect i should provide the buffer i want to release.

If we go for `spsc_pbuf_free(struct spsc_pbuf *pb)` there is less control over it. User may accidentally free twice which will result in data drop. Still, that's just my opinion.