zephyrproject-rtos/zephyr #59445: Json JSON_OBJ_DESCR_OBJ_ARRAY misbehaving

ofirshe, 2023-06-20:
Hey, I have an array of board_discovery_temp_sensor_device which utilizes the following structures:

```
typedef struct board_discovery_device_handle
{
    const char *device_description;
    uint32_t device_id;
    uintptr_t device;
} board_discovery_device_handle_t;

typedef struct health_monitor_configurations_temp_sensor
{
	int32_t over_temperature_limit;
	int32_t under_temperature_limit;
} health_monitor_configurations_temp_sensor_t;

typedef struct board_discovery_temp_sensor_channel
{
    const char *channel_name;
    uint32_t channel_id;
    health_monitor_configurations_temp_sensor_t channel_monitoring_params;
} board_discovery_temp_sensor_channel_t;

typedef struct board_discovery_temp_sensor_device
{
    board_discovery_device_handle_t device_handle;

    /* In case temp_sensor has channels, temp_sensor_monitoring_params will be unused because we should have controling params per channel. */
    health_monitor_configurations_temp_sensor_t temp_sensor_monitoring_params;

    /* In case temp_sensor has no channels, channels_arr_len will be assigned to 0 and channels_arr will be unused. */
    board_discovery_temp_sensor_channel_t channels_arr[BOARD_DISCOVERY_TEMP_SENSOR_MAX_CHANNELS];
    uint32_t channels_arr_len;
} board_discovery_temp_sensor_device_t;
```

```
Here are the coresponding json descriptions:
static const struct json_obj_descr g_board_discovery_device_handle_json_description[] = 
{
	JSON_OBJ_DESCR_PRIM(struct board_discovery_device_handle, device_description, JSON_TOK_STRING),
	JSON_OBJ_DESCR_PRIM(struct board_discovery_device_handle, device_id, JSON_TOK_NUMBER),
    JSON_OBJ_DESCR_PRIM(struct board_discovery_device_handle, device, JSON_TOK_NUMBER),
};

static const struct json_obj_descr g_board_discovery_temp_sensor_health_monitor_json_description[] =
{
	JSON_OBJ_DESCR_PRIM(struct health_monitor_configurations_temp_sensor, over_temperature_limit, JSON_TOK_NUMBER),
    JSON_OBJ_DESCR_PRIM(struct health_monitor_configurations_temp_sensor, under_temperature_limit, JSON_TOK_NUMBER),
};

static const struct json_obj_descr g_board_discovery_temp_sensor_channel_json_description[] =
{
    JSON_OBJ_DESCR_PRIM(struct board_discovery_temp_sensor_channel, channel_name, JSON_TOK_STRING),
    JSON_OBJ_DESCR_PRIM(struct board_discovery_temp_sensor_channel, channel_id, JSON_TOK_NUMBER),
    JSON_OBJ_DESCR_OBJECT(struct board_discovery_temp_sensor_channel, channel_monitoring_params, g_board_discovery_temp_sensor_health_monitor_json_description),
};

static const struct json_obj_descr g_board_discovery_temp_sensor_device_json_description[] =
{
	JSON_OBJ_DESCR_OBJECT(struct board_discovery_temp_sensor_device, device_handle, g_board_discovery_device_handle_json_description),
    JSON_OBJ_DESCR_OBJECT(struct board_discovery_temp_sensor_device, temp_sensor_monitoring_params, g_board_discovery_temp_sensor_health_monitor_json_description),
    JSON_OBJ_DESCR_OBJ_ARRAY(struct board_discovery_temp_sensor_device, channels_arr, BOARD_DISCOVERY_TEMP_SENSOR_MAX_CHANNELS, channels_arr_len,
				 g_board_discovery_temp_sensor_channel_json_description, ARRAY_SIZE(g_board_discovery_temp_sensor_channel_json_description)),
};
```

The board_discovery_temp_sensor_device array is a part of a larger structure, and here is its declaration:
```
board_discovery_temp_sensor_device_t temp_sensor_devices_arr[BOARD_DISCOVERY_TEMP_SENSOR_NUM_DEVICES];
 uint32_t temp_sensor_devices_arr_len;
```

and it's correspond json description:
/* Temperature Sensor Devices */ 
```
JSON_OBJ_DESCR_OBJ_ARRAY(struct board_discovery_devices_arr, temp_sensor_devices_arr, BOARD_DISCOVERY_TEMP_SENSOR_NUM_DEVICES,
                temp_sensor_devices_arr_len, g_board_discovery_temp_sensor_device_json_description,
                ARRAY_SIZE(g_board_discovery_temp_sensor_device_json_description)),

```

The array initialization (I have only two devices):
.temp_sensor_devices_arr = {
        [0] = {
            .device_handle = {
                .device_description = "MAX31730 Temperature Sensor",
                .device_id = 25,
                //.device = (uint32_t)DEVICE_DT_GET(DT_ALIAS(temp_sensor_0)), 
                .device = (uint32_t)NULL,
            },
            /* temp_sensor_monitoring_params variable is unused for this device because it has channels. */

            /* Channels */
            .channels_arr = {
                [0] = {
                    .channel_name = "Temperature Sensor Channel 0",
                    .channel_id = 0,
                    .channel_monitoring_params = {
                        .over_temperature_limit = 70,
                        .under_temperature_limit = 50,
                    },
                },
                [1] = {
                    .channel_name = "Temperature Sensor Channel 1",
                    .channel_id = 1,
                    .channel_monitoring_params = {
                        .over_temperature_limit = 70,
                        .under_temperature_limit = 50,
                    },
                },
                [2] = {
                    .channel_name = "Temperature Sensor Channel 2",
                    .channel_id = 2,
                    .channel_monitoring_params = {
                        .over_temperature_limit = 70,
                        .under_temperature_limit = 50,
                    },
                },
            },
            .channels_arr_len = 3,
            /* End Of Channels */
        },
        [1] = {
            .device_handle = {
                .device_description = "Protean-tech Temperature Sensor",
                .device_id = 26,
                //.device = (uint32_t)DEVICE_DT_GET(DT_ALIAS(temp_sensor_1)), 
                .device = (uint32_t)NULL,
            },
            .temp_sensor_monitoring_params = {
                .over_temperature_limit = 70,
                .under_temperature_limit = 50,
            },
            /* channels_arr variable is unused for this device because it has no channels. */
            .channels_arr_len = 0,
        },
    },
    .temp_sensor_devices_arr_len = 0,

When I print the output json of this array it looks like this:
```
"temp_sensor_devices_arr":[{"device_handle":{"device_description":"MAX31730 Temperature Sensor","device_id":9,"device":0},"temp_sensor_monitoring_params":{"over_temperature_limit":0,"under_temperature_limit":0},"channels_arr":[{"channel_name":"Temperature Sensor Channel 0","channel_id":0,"channel_monitoring_params":{"over_temperature_limit":70,"under_temperature_limit":50}},{"channel_name":"Temperature Sensor Channel 1","channel_id":1,"channel_monitoring_params":{"over_temperature_limit":70,"under_temperature_limit":50}},{"channel_name":"Temperature Sensor Channel 2","channel_id":2,"channel_monitoring_params":{"over_temperature_limit":70,"under_temperature_limit":50}}]},
                           {"device_handle":{"device_description":"O\\fF\x16F\x07i\\nFD\xea\x06\\b\x87\xb0\x05F\x97\xf8$\x10\xc8\xf3\x0b\\b8h","device_id":1610754939,"device":10},"temp_sensor_monitoring_params":{"over_temperature_limit":0,"under_temperature_limit":70},"channels_arr":[]}], 
```

While debugging, I observed that after finishing the encoding of temp_sensor_devices_arr[0], the val pointer in the encode function advanced to the offset of the channels_arr_len field. This explains why, when inspecting the json_escape_internal() function in the debugger, the value of the str pointer is 0x3. I also verified against the actual array, and it corresponds to the address of channels_arr_len from element 0. However, it should actually be the start address of temp_sensor_devices_arr[1]. The channels_arr_len field should be encapsulated within the call to JSON_OBJ_DESCR_OBJ_ARRAY, and it should be taken into account when calculating the offset of each object.
[image]

Zephyr version: 3.4.99 
build: v3.4.0-rc1-723-gf08ba0d654ee

Thanks, 
Ofir.

henrikbrixandersen, 2023-06-20:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

ofirshe, 2023-06-20:
Upon further observation, I noticed that there is an open PR that could potentially shed light on the issue I'm currently facing. It seems that the PR addresses a similar scenario involving nested arrays, which aligns with my situation.  https://github.com/zephyrproject-rtos/zephyr/pull/50816/files#diff-d60199059b33fa2cc46a73a35a5f8cc5569238277adc4662858743af0fd487fa

ofirshe, 2023-06-26:
Hey, 

Any update on this matter?

mrfuchs, 2023-06-30:
Fixed by #50816 which was merged in ae6aa61.

bbilas, 2023-07-05:
@mrfuchs 

I'm reopening that issue since there are still troubles while multidimensional array encoding. https://github.com/zephyrproject-rtos/zephyr/pull/50816 fixes the case when there is  declared `struct json_obj_descr` with only one multidimensional array but when you add another field then there is no data in that array. I've used the code from the tests that you have written to show the problem.

The encoding works well when I declare the following `struct json_obj_descr`
```c
static const struct json_obj_descr test_multidimensional_array_descr[] = {
	JSON_OBJ_DESCR_ARRAY_ARRAY(struct obj_array_2dim, data, 4,
				   objects_array_array_len, obj_array_descr,
				   ARRAY_SIZE(obj_array_descr)),
};
```

with the test data:

```c
struct obj_array_2dim obj_array_array_ts = {
		.data = {
			[0] = {
				.data = {
					[0] = {
						.name = "Sim\303\263n Bol\303\255var",
						.height = 168
					},
					[1] = {
						.name = "Pel\303\251",
						.height = 173
					},
					[2] = {
						.name = "Usain Bolt",
						.height = 195
					},
				},
				.num_elements = 3
			},
			[1] = {
				.data = {
					[0] = {
						.name = "Muggsy Bogues",
						.height = 160
					},
					[1] = {
						.name = "Hakeem Olajuwon",
						.height = 213
					},
				},
				.num_elements = 2
			},
			[2] = {
				.data = {
					[0] = {
						.name = "Alex Honnold",
						.height = 180
					},
					[1] = {
						.name = "Hazel Findlay",
						.height = 157
					},
					[2] = {
						.name = "Daila Ojeda",
						.height = 158
					},
					[3] = {
						.name = "Albert Einstein",
						.height = 172
					},
				},
				.num_elements = 4
			},
		},
		.objects_array_array_len = 3,
};
```

I have the correct output:

```
[00:00:01.335,449] <inf> cloud: json data {"data":[[{"name":"Simón Bolívar","height":168},{"name":"Pelé","height
":173},{"name":"Usain Bolt","height":195}],[{"name":"Muggsy Bogues","height":160},{"name":"Hakeem Olajuwon","height":
213}],[{"name":"Alex Honnold","height":180},{"name":"Hazel Findlay","height":157},{"name":"Daila Ojeda","height":158}
,{"name":"Albert Einstein","height":172}]]}
```

but when I add another field e.g
```c
static const struct json_obj_descr test_multidimensional_array_descr[] = {
	JSON_OBJ_DESCR_PRIM(struct test_string_struct, tId, JSON_TOK_STRING),
	JSON_OBJ_DESCR_ARRAY_ARRAY(struct obj_array_2dim, data, 4,
				   objects_array_array_len, obj_array_descr,
				   ARRAY_SIZE(obj_array_descr)),
};
```

then in the multidimensional array, there is no data.

```
[00:00:01.335,723] <inf> cloud: json data {"tId":"TEST1","data":[]}
```

Do you have any clues what could be wrong?

closed 2023-07-05 by bbilas