zephyrproject-rtos/zephyr #59453: `TOOLCHAIN_HOME` should be settable via an cmake variable.

aaronemassey, 2023-06-20:
**Is your enhancement proposal related to a problem? Please describe.**
`TOOLCHAIN_HOME` should be settable via an cmake variable.

**Describe the solution you'd like**
Check if `TOOLCHAIN_HOME` has been defined as a cmake variable, and if not check if it's an environment variable.

**Describe alternatives you've considered**
Do nothing. `TOOLCHAIN_HOME` remains as settable only as an environment variable.

**Additional context**

Have a good idea of what to do here, but wanted to create a gh issue to track it.

aaronemassey, 2023-06-20:
Nevermind, I was mistaken.

aaronemassey, 2023-06-20:
Reopening this issue for `ZEPHYR_TOOLCHAIN_VARIANT` only being available to `verify-toolchain.cmake` as an environment variable.

aaronemassey, 2023-06-20:
I could probably tackle this, but unassigned myself in case someone else wants to pick it up.

tejlmand, 2023-08-24:
> Reopening this issue for `ZEPHYR_TOOLCHAIN_VARIANT` only being available to `verify-toolchain.cmake` as an environment variable.

`verify-toolchain.cmake` can handle `ZEPHYR_TOOLCHAIN_VARIANT` both as a CMake variable and as a an environment variable.

For example when `ZEPHYR_TOOLCHAIN_VARIANT` is not given as CMake variable, then the environment is checked:
```fish
$ echo $ZEPHYR_TOOLCHAIN_VARIANT
llvm
$ cmake -DFORMAT=json -P ./cmake/verify-toolchain.cmake
{"ZEPHYR_TOOLCHAIN_VARIANT" : "llvm", "SDK_VERSION": "", "ZEPHYR_SDK_INSTALL_DIR" : ""}
```

but if given as CMake variable, then that is used instead:
```fish
$ echo $ZEPHYR_TOOLCHAIN_VARIANT
llvm
$ cmake -DFORMAT=json -DZEPHYR_TOOLCHAIN_VARIANT=zephyr -P ./cmake/verify-toolchain.cmake
{"ZEPHYR_TOOLCHAIN_VARIANT" : "zephyr", "SDK_VERSION": "0.16.1", "ZEPHYR_SDK_INSTALL_DIR" : "/opt/zephyr-sdk-0.16.1"}
```

i'm unsure what you mean by this:
> ZEPHYR_TOOLCHAIN_VARIANT should be viewable as cmake variable.

I can read / use ZEPHYR_TOOLCHAIN_VARIANT as a regular CMake variable in a custom CMake file, for example create the following file:
```
# custom_toolchain_script.cmake
cmake_minimum_required(VERSION 3.20)

include(${CMAKE_CURRENT_LIST_DIR}/cmake/verify-toolchain.cmake)
message("ZEPHYR_TOOLCHAIN_VARIANT: ${ZEPHYR_TOOLCHAIN_VARIANT}")
```
and then run the script as:
```
$ echo $ZEPHYR_TOOLCHAIN_VARIANT
llvm

$ cmake -P ./custom_toolchain_script.cmake
...
ZEPHYR_TOOLCHAIN_VARIANT: llvm

$ cmake -DZEPHYR_TOOLCHAIN_VARIANT=zephyr -P ./custom_toolchain_script.cmake
...
ZEPHYR_TOOLCHAIN_VARIANT: zephyr
```

so the toolchain variant is available as a CMake var to read, when using `verify-toolchain.cmake`.

Or @aaronemassey are youactually requisting the `twister` itself can take the toolchain as argument ?

`twister` itself uses the Zephyr toolchain variant from environment, and has always done so.
Something like `./scripts/twister --toolchain="llvm" ....` 

Cause then I believe the enhancement issue should be clarified and turned into an enhancement regarding twister.