zephyrproject-rtos/zephyr #59457: WebUSB sample not recognized correctly in Windows 10

maxd-nordic, 2023-06-20:
**Describe the bug**
Automatic driver installation does not work for the WebUSB sample. It just gets assigned the null GUID and is shown with a warning triangle. Zadig also shows that the device did not get the correct WICD. On Linux, the WebUSB stuff seems to work fine, so I presume this is mostly related to the MS OS 2.0 descriptors / BOS entry. The MS OS 1.0 stuff seems to be irrelevant for anything newer than Windows 7.

**To Reproduce**
 - Build the WebUSB sample with an nRF52 or nRF53 DK. (I don't think this is hardware-related.)
 - Plug into a Windows 10 machine or VM

**Expected behavior**
The WinUSB driver should be installed automatically, the device should not be should with a warning symbol in the device manager. The GUID should be reported correctly.

**Impact**
This is a showstopper for me since it prevents me from implementing my current project: A Zephyr implementation of CMSIS-DAP v2. The automatic driver installation part is mandatory to conform to the spec.

**Environment (please complete the following information):**
 - Windows 10
 - Zephyr 3.3.0

carlescufi, 2023-06-21:
> This is a showstopper for me since it prevents me from implementing my current project: A Zephyr implementation of CMSIS-DAP v2. The automatic driver installation part is mandatory to conform to the spec.

Have you seen this? https://github.com/zephyrproject-rtos/zephyr/pull/53798

maxd-nordic, 2023-06-21:
Yes, I'm trying to finish that one for Johann. :)

tmon-nordic, 2023-06-22:
The problem is definitely in MS OS 2.0 descriptor. One of the issue is that
https://github.com/zephyrproject-rtos/zephyr/blob/c6c93e394d6a52c4541d39135681fc81b8df7dd5/samples/subsys/usb/webusb/src/main.c#L40
hardcodes the first interface index to 0x02 while the sample when built for nrf52840dk_nrf52840 has only one interface with number 0.

Changing the index itself doesn't seem to be enough though.

closed 2023-08-28 by maxd-nordic