zephyrproject-rtos/zephyr #59459: Bluetooth: hci: spi: host doesn't work for targets which do not supporting level triggered interrupts

lucasdietrich, 2023-06-20:
**Bug Description**
An application implementing an HCI HOST will constantly fail on targets that do not support GPIO level-triggered interrupts.

This change from edge to level-triggered interrupts was introduced in the following PR #57427 which addresses issue #53980.

**Steps for Reproduction**
To replicate the issue, build a Bluetooth sample configure as a HCI HOST (i.e. with `CONFIG_BT_HCI=y`, `CONFIG_BT_SPI=y` and `CONFIG_BT_CTLR=n`) on a board that doesn't support level-triggered GPIO interrupts (e.g., any stm32 board, stm32f429zi-nucleo in my particular instance).

The problem arises when the host can't configure the IRQ pin as a level-triggered interrupt via [`gpio_pin_interrupt_configure_dt(&irq_gpio, GPIO_INT_LEVEL_ACTIVE);` ](https://github.com/zephyrproject-rtos/zephyr/blob/356c8cbe63ae01b3ab438382639d25bb418a0213/drivers/bluetooth/hci/spi.c#L309).

The stm32 driver explicitly tells level-triggered interrupts are not supported [here](https://github.com/zephyrproject-rtos/zephyr/blob/732dd4843dc5c21f83f17d97bbafbfad8a3bf91a/drivers/gpio/gpio_stm32.c#L614).

**Expected Result**
HCI HOST should ideally be functional even with targets that don't support level-triggered interrupts.

**Consequences**
As it stands, the HCI HOST is incompatible with targets that don't support level-triggered interrupts.

**Environment Details:**
- Linux Fedora 37
- Zephyr v3.4.0
- Zephyr SDK 0.16.1

**Additional Information**
I imagine that only few targets support level-triggered interrupts.
Consequently, it might be more effective to design the HCI driver with the assumption that only edge-triggered interrupts are supported.

While I haven't yet formulated a resolution, I plan to think about it.

jori-nordic, 2023-06-21:
tagging @JordanYates since he did the original change.

JordanYates, 2023-06-21:
I appreciate the issue, but simply reverting the change just re-introduces the original issue.
Presumably this sort of issue has been encountered before. Is the standard solution to only use lowest common denominator features in drivers or to introduce some form of emulation for chips missing the feature?
I note that `GPIO_INT_LEVEL_ACTIVE` is used in a few other sensor drivers, which would be affected by the same problem.

erwango, 2023-06-22:
> Is the standard solution to only use lowest common denominator features in drivers or to introduce some form of emulation for chips missing the feature?

I would opt for "lowest common denominator features".
Alternative would mean each driver as to provide emulation for any missing feature. Not sure this would scale.

JordanYates, 2023-06-22:
> I would opt for "lowest common denominator features".
Alternative would mean each driver as to provide emulation for any missing feature. Not sure this would scale.

The same argument also applies to lowest common denominator. If Zephyr starts supporting some crazy micro that only supports level triggered interrupts, where does that leave us?

JordanYates, 2023-06-22:
It's also not really feasible to expect driver developers to check each implementation of peripherals they use to see what is and isn't supported. My first expectation is that if its in a public header like `zephyr/drivers/gpio.h` then it's fair game to use, and failure to support such a feature is more a problem for the micro implementation than the device driver.

erwango, 2023-06-22:
> The same argument also applies to lowest common denominator. If Zephyr starts supporting some crazy micro that only supports level triggered interrupts, where does that leave us?

Fair point.

> and failure to support such a feature is more a problem for the micro implementation than the device driver.

This doesn't justify in any case to break a previously working functionality.

jori-nordic, 2023-06-22:
> This doesn't justify in any case to break a previously working functionality.

I think 'previously working' is up for debate, since @JordanYates did fix a bug.

We're planning to put the SPI and UART (H4) drivers in regression at Nordic. But this will only be done for Nordic chips, so if other members of the community / SoC vendors could do the same with their hardware, that would be great. 

That way anyone can NAK the problematic PRs before the functionality is broken (I agree, it's not great) and we can try to come up with a solution that satisfies everyone.

A week of delay for those driver PRs would seem like a fair amount of time for those tests to be run, either in CI or manually. Passed that, I think it's ok to merge.

erwango, 2023-06-22:
> I think 'previously working' is up for debate, since @JordanYates did fix a bug.

According to current PR description, it was at least working on some platforms and there were some race conditions on some others. Now it appears to be fully broken where it was previously working.

I don't blame anyone to introduce regressions, it can happen as it's hard to figure out all possible use cases and configurations related to a piece of code, unless informed people provide feedback during the review, which obviously didn't happen here.
Though, I regret the position which states "ok, this is is because of missing HW functionality vs the new implementation, now deal with it".

For similar case, in general the commit introducing the regression is reverted and the point is reworked in order to provide a satisfying solution for all known use cases and configurations.

erwango, 2023-06-22:
Note that I'm counting 16 gpio drivers stating `GPIO_INT_MODE_LEVEL` is not supported.

JordanYates, 2023-06-22:
> According to current PR description, it was at least working on some platforms and there were some race conditions on some others. Now it appears to be fully broken where it was previously working.

It would work for some duration, until it deadlocked and never worked again. This was for all platforms, not just some.

> For similar case, in general the commit introducing the regression is reverted and the point is reworked in order to provide a satisfying solution for all known use cases and configurations.

The fix is logically correct. If the IRQ line is active, we should be attempting to read the data. Triggering on an edge is insufficient and will eventually lead to a deadlock.

Assuming we take the suggested approach, the only "satisfying solution" is one in which equivalent behaviour to `GPIO_INT_LEVEL_ACTIVE` is obtained through some combination of `GPIO_INT_EDGE_ACTIVE` and `gpio_pin_read`. Assuming that the following is possible, why not instead implement it in a common software emulation layer that implementations can call if they don't support `GPIO_INT_MODE_LEVEL`? The implementation is the same, and it benefits all drivers instead of just this.

Otherwise we might as well add a Github action that fails any PR that uses `GPIO_INT_LEVEL_ACTIVE` outside of the `boards` folder.

JordanYates, 2023-06-22:
> We're planning to put the SPI and UART (H4) drivers in regression at Nordic. But this will only be done for Nordic chips, so if other members of the community / SoC vendors could do the same with their hardware, that would be great.

What are you planning to transition to, if you are able to share that?

erwango, 2023-06-22:
> It would work for some duration, until it deadlocked and never worked again. This was for all platforms, not just some.

No deadlock to fear anymore indeed.

> Assuming we take the suggested approach, the only "satisfying solution" is one in which equivalent behaviour to GPIO_INT_LEVEL_ACTIVE is obtained through some combination of GPIO_INT_EDGE_ACTIVE and [...]

In the light of previous discussion, this approach looks ok indeed.
But (and I may appear stubborn), I persist in stating that https://github.com/zephyrproject-rtos/zephyr/pull/57427 should be reverted first, as previous status looked overall better than current one.

@lucasdietrich did you ever faced https://github.com/zephyrproject-rtos/zephyr/issues/53980 and if yes was it blocking ?

JordanYates, 2023-06-23:
> But (and I may appear stubborn), I persist in stating that https://github.com/zephyrproject-rtos/zephyr/pull/57427 should be reverted first, as previous status looked overall better than current one.

That is a fair approach to take, my concern is where the responsibility then lies for fixing the issue.
It's easy for me to justify spending time fixing Bluetooth bugs, less so writing hardware support for platforms I don't use and don't have access to.
It seems likely the end result will be this driver remaining subtly broken.

JordanYates, 2023-06-23:
As this conversation is larger than this bug report, I opened a discussion: https://github.com/zephyrproject-rtos/zephyr/discussions/59646

jori-nordic, 2023-06-23:
> What are you planning to transition to, if you are able to share that?

Maybe I was not clear, I meant "in regression" as in regression testing/CI. We're not deprecating anything.

We already have some testing on real hardware in nRF connect SDK, the plan is to modify that framework a little bit to use two devices where there was only one: one running e.g. the hci_spi sample, and the other running the testing app + the host with the SPI driver. We'll then run the same host test suite on that setup.

If this proves too complicated, I'll at the very least cobble a test setup that does the same, just with two bt samples (e.g. central_hr and peripheral_hr). Other ideas are the shell or PTS' BTP protocol, dunno yet.
I know the uart H4 driver is tested as a byproduct of controller qualification, but AFAIK the SPI driver is not tested anywhere.

Now that I think about it, a small framework + some simple tests based on the BTP protocol could be merged upstream, allowing for easy testing by other silicon vendors. Don't know if I can afford the time to work on that right now though. My understanding is the BTP protocol wasn't designed for performance testing, rather qualification. But it is extensible.

coran21, 2023-06-28:
I updated my Zephyr to 3.4 and the fact that the HCI HOST (SPI) stopped working brought me here. 
I'm developing the project based on STM32 and HCI HOST (SPI) and so far never run into deadlock problem (working with bluetooth for couple months), but I believe that it can happen.

I'm relatively new to Zephyr so I'm sorry if what I write below doesn't make any sense.

I'm trying to find some temporary workaround for this and it's clear that I have to use previous version of the file which works with edge triggered interrupts because my STM32 doesn't support level triggered ones.
What would happen if we just don't disable IRQ interrupt at all? 
There could be some additional calls to the interrupt callback function, but that just set the semaphore to maximum limit 1 and it would process the rx thread loop one more time.
Would there be big performance impact? 
It's possible that I'm missing something.

I also tested the code with `gpio_pin_get` every 1s which works fine, but it's little bit weird.

lucasdietrich, 2023-06-29:
> I'm trying to find some temporary workaround for this and it's clear that I have to use previous version of the file which works with edge triggered interrupts because my STM32 doesn't support level triggered ones.

As I've never faced #53980 too, I've personally just `git revert` commit ca6c08f960d83265497af261c5eda2e782c47b26 on a fork of Zephyr as a temporary fix.

erwango, 2023-07-05:
@JordanYates Are you fine with reverting https://github.com/zephyrproject-rtos/zephyr/pull/57427 :) ?

coran21, 2023-07-12:
Sorry for bringing that again, but what would be the worst case scenario when the IRQ interrupt won't be disabled at all? I'm currently testing this as a workaround and I didn't run into any problem yet, but as I said in last comment it's possible that I'm missing something.

JordanYates, 2023-07-12:
> Sorry for bringing that again, but what would be the worst case scenario when the IRQ interrupt won't be disabled at all? I'm currently testing this as a workaround and I didn't run into any problem yet, but as I said in last comment it's possible that I'm missing something.

Assuming there was a misbehaving controller device which sometimes pulls the INT line low while data is being read (which is what the old logic looks to be guarding against), the host will enter a pretty busy loop until the controller DOES have data to send, which may be soon or may be never. This could be worked around with a timeout on our side of course.

closed 2023-09-01 by carlescufi

arbrauns, 2023-10-20:
Dang, I missed this - unfortunately we have the opposite problem, the HCI interrupt pin is routed through an MCP23xxx port expander, which doesn't reliably support edge-triggered interrupts (level-triggered interrupts are also broken, but at least they just generate *too many* interrupts, which can be dealt with). Guess I'll revert this in our fork and hope it doesn't become too much of a mess in the future.

JordanYates, 2023-10-20:
Given that I would prefer to continue using the level triggered interrupts as well, I might look into how to support both.

PR #60389 (2023-09-01): Bluetooth: hci: spi: Use edge sensitive interrupt and mutex fixes.
Revert to edge sensitive interrupt. (Fixes #59459 )
Optimized BlueNRG-MS access.
Addressed issue for mutex access of SPI.
Fixed chip select active state for x_nucleo_idb05a1.
I also tested samples/bluetooth/beacon on 96b_carbon board and it works fine; however, without my modifications in spi.c file the application crashes since ST boards do not support level trigger interrupt.