zephyrproject-rtos/zephyr #59461: samples: drivers: led_lp5569: sample.drivers.led.lp5569 on nrf52840dk_nrf52840 fails

katgiadla, 2023-06-21:
**Describe the bug**
The samples/drivers/led_lp5569/sample.drivers.led.lp5569 fails.

Observed for

* nrf52840dk_nrf52840

**To Reproduce**
Steps to reproduce the behavior:

1. have `nrf52840dk_nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T samples/drivers/led_lp5569/ -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-06-21T00:19:08.957Z] [0mERROR   - [00:00:00.882,720] <err> i2c_nrfx_twi: Error on I2C line occurred for message 0[0m
[2023-06-21T00:19:08.957Z] [0m[00:00:00.882,781] <err> lp5569: Enable LP5569 failed[0m
[2023-06-21T00:19:08.957Z] [0m*** Booting Zephyr OS build zephyr-v3.4.0-168-gdf0e03c5512c ***[0m
[2023-06-21T00:19:08.957Z] [0m[00:00:00.882,873] <err> app: LED device lp5569@32 not ready[0m
[2023-06-21T00:19:08.957Z] [0m[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr 0.16.0
 - Commit SHA or Version used: zephyr-v3.4.0-168-gdf0e03c5512c

nordicjm, 2023-06-21:
Were you testing this with a shield? Because I don't believe the nrf52840dk has such hardware on the board itself, so the test *should* fail.

nordicjm, 2023-06-21:
Yes as per the files added in https://github.com/zephyrproject-rtos/zephyr/commit/22e7edbbab1d88209cc1ac664f7f853956041f1a it seems odd that this was added with an nrf52840dk sample when the documentation has:

> Build the application for the :ref:`nrf52840dk_nrf52840` board, and connect
> a LP5569 LED controller on the bus I2C0 at the address 0x32.

katgiadla, 2023-06-26:
Hardware were not connected