zephyrproject-rtos/zephyr #59473: Create a repository for Ambiq HAL

msobkowski, 2023-06-21:
## Origin

The HAL for the Ambiq SoCs is provided as a part of the AmbiqSuite SDK as an downloadable archive.
It can be downloaded from the following location: https://ambiq.com/ambiqsuite-sdk-4-3-0.

## Purpose

The HAL is used in SoC support for the Apollo4 SoC series. It is used for platform initialization and device drivers.

## Mode of integration

The module should be integrated as an external HAL module. 

The proposed name is hal_ambiq. It will contain only the HAL code (subset of the AmbiqSuite SDK).
The proposed tree layout is as follows:
```
.
├── boards/
├── CMakeLists.txt
├── CMSIS/
├── devices/
├── mcu/
├── utils/
└── zephyr
    └── module.yml
```

## Maintainership

Ambiq HAL will be maintained by the following people:
- maintainer: @fkokosinski (Antmicro)
- collaborator: @tgorochowik (Antmiro)
- collaborator: @msobkowski (Antmicro)

## Pull Request

Intended content of the Ambiq HAL repository can be found [here](https://github.com/antmicro/hal_ambiq/tree/main).

Pull request adding support for Ambiq Apollo4 Plus SoC that uses this module: https://github.com/zephyrproject-rtos/zephyr/pull/59470

## Description

AmbiqSuite SDK is a C software development kit provided by Ambiq for their line of MCUs (the latest release, v4.3.0, supports Apollo4, Apollo4 Blue, Apollo4 Plus, and Apollo4 Blue Plus SoC based EVBs). The HAL is standalone and OS agnostic. It provides the device headers, device drivers and build system for writing programs for the Ambiq SoCs.

The AmbiqSutite SDK package besides the HAL itself, also provides sample applications and integrates with 3rd party software. Those parts will not be part of the Ambiq HAL in Zephyr, as they are not required and would introduce licensing issues.

## Dependencies

None

## Revision

AmbiqSuite SDK 4.3.0

## License

``BSD-3-Clause`` - HAL files
``Apache-2.0`` - CMSIS headers, CMake files

galak, 2023-07-12:
@carlescufi have we normally asked for an example repo for the HAL in cases that there isn't an upstream?  Thus we can truly review the HAL, licenses, etc.

nashif, 2023-07-12:
> @carlescufi have we normally asked for an example repo for the HAL in cases that there isn't an upstream? Thus we can truly review the HAL, licenses, etc.

https://github.com/antmicro/hal_ambiq/tree/main ?

galak, 2023-07-12:
Remove the CMSIS dir and board, the board code should be from the zephyr tree itself.

kgugala, 2023-07-13:
> Remove the CMSIS dir and board, the board code should be from the zephyr tree itself.

This is done. We also dropped the `Boards` directory (as discussed in the TSC meeting)

closed 2023-07-25 by nashif