zephyrproject-rtos/zephyr #59506: [Coverity CID: 321155] Untrusted loop bound in arch/x86/core/acpi.c

zephyrbot, 2023-06-21:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/ce3317d03e46e1a218af3e712383d9216c48a992/arch/x86/core/acpi.c

Category: Insecure data handling
Function: `z_acpi_get_cpu`
Component: Architectures
CID: [321155](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=321155)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/ce3317d03e46e1a218af3e712383d9216c48a992/arch/x86/core/acpi.c#L260

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_TAINTED_SCALAR). ([CWE-606](http://cwe.mitre.org/data/definitions/606.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

fabiobaltieri, 2023-08-22:
@jhedberg could you look into this or reassign it?

jhedberg, 2023-08-28:
@najumon1980 could you take a look at this?

jhedberg, 2023-08-28:
@tbursztyka also FYI - not sure if this is new ACPICA related code or existing ACPI code

najumon1980, 2023-08-28:
This looks existing code and will be replaced with the new acpica interface PR: https://github.com/zephyrproject-rtos/zephyr/pull/59917

jhedberg, 2023-11-07:
The file (and code) has been removed from the Zephyr tree.

closed 2023-11-07 by jhedberg