zephyrproject-rtos/zephyr #59507: [Coverity CID: 248391] Out-of-bounds access in subsys/bluetooth/shell/ll.c

zephyrbot, 2023-06-21:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/ce3317d03e46e1a218af3e712383d9216c48a992/subsys/bluetooth/shell/ll.c

Category: Memory - corruptions
Function: `cmd_advx`
Component: Bluetooth
CID: [248391](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=248391)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/ce3317d03e46e1a218af3e712383d9216c48a992/subsys/bluetooth/shell/ll.c#L253

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cvinayak, 2023-07-01:
<html><body>
<h4><span ng-switch="" on="changeItem.fromCommit" class="ng-binding">2023-07-01 00:25:42<span ng-switch-default="" class="ng-binding ng-scope"></span></span></h4>

Classification: | Unclassified | → | Bug
-- | -- | -- | --
Severity: | Unspecified | → | Minor
Action: | Undecided | → | Fix Required

</body>
</html>

cvinayak, 2023-07-06:
<html><body>
<h4><span ng-switch="" on="changeItem.fromCommit" class="ng-binding">2023-07-06 23:03:22<span ng-switch-default="" class="ng-binding ng-scope"> (change by vinayak.kariappa.chettimada@nordicsemi.no)</span></span></h4>

Classification: | Bug | → | False Positive
-- | -- | -- | --
Action: | Fix Required | → | Ignore

<h5 ng-show="changeItem.comment" class="triage-comment ng-binding">Comment:</h5><p ng-show="changeItem.comment" class="triage-comment ng-binding">False positive, cmd_advx shall not be compiled-in when CONFIG_BT_CTLR_ADV_EXT=n. And evt_prop will not cause an overrun.</p>
</body>
</html>

closed 2023-07-06 by cvinayak