zephyrproject-rtos/zephyr #59509: [Coverity CID: 318800] Untrusted loop bound in subsys/bluetooth/controller/ll_sw/nordic/lll/lll_central_iso.c

zephyrbot, 2023-06-21:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/ce3317d03e46e1a218af3e712383d9216c48a992/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_central_iso.c#L978

Category: Insecure data handling
Function: `isr_rx`
Component: Bluetooth
CID: [318800](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=318800)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/ce3317d03e46e1a218af3e712383d9216c48a992/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_central_iso.c#L978

```
972              param = next_cis_lll;
973              next_chan_use = next_cis_chan;
974              data_chan_prn_s = next_cis_chan_prn_s;
975              data_chan_remap_idx = next_cis_chan_remap_idx;
976      }
977     
>>>     CID 318800:  Insecure data handling  (TAINTED_SCALAR)
>>>     Passing tainted expression "param->event_count" to "isr_prepare_subevent", which uses it as a loop boundary.
978      isr_prepare_subevent(param);
979     
980      return;
981     
982     isr_rx_done:
983      radio_isr_set(isr_done, param);
54 new defect(s) introduced to Zephyr found with Coverity Scan.
238 defect(s), reported by Coverity Scan earlier, were marked fixed in the recent build analyzed by Coverity Scan.
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_TAINTED_SCALAR). ([CWE-606](http://cwe.mitre.org/data/definitions/606.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

fabiobaltieri, 2023-08-22:
@cvinayak hey could you look into this?

nashif, 2023-12-21:
> Looks like the zephyrbot accidentally reopened this. The CID in coverity is still closed, and the issue should have been fixed.

coverity still thinks the issue is there...

Thalley, 2023-12-21:
> > Looks like the zephyrbot accidentally reopened this. The CID in coverity is still closed, and the issue should have been fixed.
> 
> coverity still thinks the issue is there...

It was marked as "Unclassified" - Maybe that was the issue? 
I have changed it to be a bug, as it has been resolved. 

If the issue still persists, shouldn't coverity, after the initial issue has been marked as resolved, create a new CID?

Thalley, 2023-12-22:
Since I just changed the state of this issue in Coverity (just before the script was re-run), I'll close again and see if it appears again next time.

closed 2025-02-11 by kartben