zephyrproject-rtos/zephyr #59532: [Coverity CID: 210031] Dereference before null check in drivers/modem/gsm_ppp.c

zephyrbot, 2023-06-21:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/ce3317d03e46e1a218af3e712383d9216c48a992/drivers/modem/gsm_ppp.c

Category: Null pointer dereferences
Function: `set_ppp_carrier_on`
Component: Drivers
CID: [210031](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210031)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/ce3317d03e46e1a218af3e712383d9216c48a992/drivers/modem/gsm_ppp.c#L622

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_REVERSE_INULL). ([CWE-476](http://cwe.mitre.org/data/definitions/476.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

str4t0m, 2023-07-09:
Fixed in PR  #59369

str4t0m, 2023-07-09:
Fixed in PR  #59369

closed 2023-07-09 by str4t0m