zephyrproject-rtos/zephyr #59537: [Coverity CID: 321117] Use of 32-bit time_t in drivers/counter/maxim_ds3231.c

zephyrbot, 2023-06-21:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/ce3317d03e46e1a218af3e712383d9216c48a992/drivers/counter/maxim_ds3231.c

Category: High impact quality
Function: `ds3231_counter_get_value`
Component: Drivers
CID: [321117](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=321117)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/ce3317d03e46e1a218af3e712383d9216c48a992/drivers/counter/maxim_ds3231.c#L767

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_Y2K38_SAFETY). ([CWE-197](http://cwe.mitre.org/data/definitions/197.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

fabiobaltieri, 2023-08-22:
@pabigot could you take a stab at this? Few warnings on this file: https://github.com/zephyrproject-rtos/zephyr/issues/59543 https://github.com/zephyrproject-rtos/zephyr/issues/59545 https://github.com/zephyrproject-rtos/zephyr/issues/59546

ceolin, 2024-01-31:
reported in latest snapshot. Priority changed to match coverity.

PR #68794 (2024-02-13): drivers: counter: maxim_ds3231: Fix coverity issues
maxim_ds3231 driver stores the current time in a 64-bit time_t variable before casting it to an unsigned 32-bit integer, resulting in CWE-197 numeric truncation violations. Since counter API cannot return 64-bit values, work around these violations by explicitly performing the truncations.

Fixes #59546
Fixes #59545
Fixes #59543
Fixes #59542
Fixes #59537
closed 2024-02-13 by henrikbrixandersen