zephyrproject-rtos/zephyr #59549: [Coverity CID: 318806] Inequality comparison against NULL in kernel/mmu.c

zephyrbot, 2023-06-21:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/ce3317d03e46e1a218af3e712383d9216c48a992/kernel/mmu.c#L746

Category: Incorrect expression
Function: `z_phys_map`
Component: Kernel
CID: [318806](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=318806)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/ce3317d03e46e1a218af3e712383d9216c48a992/kernel/mmu.c#L746

```
740      align_boundary = arch_virt_region_align(aligned_phys, aligned_size);
741     
742      key = k_spin_lock(&z_mm_lock);
743      if (flags & K_MEM_DIRECT_MAP) {
744              dest_addr = (uint8_t *)aligned_phys;
745              /* Reserve from the virtual memory space */
>>>     CID 318806:  Incorrect expression  (BAD_COMPARE)
>>>     Comparing pointer "dest_addr + aligned_size" against "NULL" using anything besides "==" or "!=" is likely to be incorrect.
746              if (!(dest_addr + aligned_size < Z_VIRT_RAM_START ||
747                  dest_addr > Z_VIRT_RAM_END)) {
748                      num_bits = aligned_size / CONFIG_MMU_PAGE_SIZE;
749                      offset = virt_to_bitmap_offset(dest_addr, aligned_size);
750                      if (sys_bitarray_test_and_set_region(
751                          &virt_region_bitmap, num_bits, offset, true))
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_BAD_COMPARE). ([CWE-480](http://cwe.mitre.org/data/definitions/480.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

dcpleung, 2023-11-16:
Why... @zephyrbot... why are you reopening this when it has been fixed... Why!!!!!

closed 2023-11-16 by dcpleung

dcpleung, 2023-11-16:
And this is comparing pointer addresses and DEFINITELY NOT trying to see if a pointer is NULL or not.

peter-mitsis, 2023-11-21:
I had a similar issue assigned to me by zephyrbot the other day as well. (See #65375).

I think that it is complaining about the use of the '<' and '>' comparisons in conjunction with the 'uint8_t *' type. I suspect that it would prefer the use of the `uintptr_t` type for the comparisons instead.

dcpleung, 2023-11-21:
Probably that's the case. However, it is still a false positive.

PR #60853 (2023-08-15): kernel: mmu: fix bitmap set and clear under direct map
When CONFIG_KERNEL_DIRECT_MAP enabled, the region to be mapped or unmapped can be outside of the virtual memory space, wholly within it, or overlap partially. Additional processing is needed to make sure we only manipulate the bits within the bitmap, in other words, only the pages represented by the bitmap.

Fixes #59549
commit b8e0de2a (2023-08-15): kernel: mmu: fix bitmap set and clear under direct map

When CONFIG_KERNEL_DIRECT_MAP enabled, the region to be mapped
or unmapped can be outside of the virtual memory space, wholly
within it, or overlap partially. Additional processing is
needed to make sure we only manipulate the bits within
the bitmap, in other words, only the pages represented by
the bitmap.

Fixes #59549