zephyrproject-rtos/zephyr #59556: [Coverity CID: 20413] Out-of-bounds read in lib/posix/getopt/getopt_long.c

zephyrbot, 2023-06-21:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/ce3317d03e46e1a218af3e712383d9216c48a992/lib/posix/getopt/getopt_long.c

Category: Memory - illegal accesses
Function: `getopt_internal`
Component: Other
CID: [20413](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=20413)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/ce3317d03e46e1a218af3e712383d9216c48a992/lib/posix/getopt/getopt_long.c#L498

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-125](http://cwe.mitre.org/data/definitions/125.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cfriedt, 2023-07-07:
@jakub-uC - just in case you've not logged in to coverity before, you should be able to sign on with github.

In any case though, I looked at this issue, and to me it seemed like it might have been a false positive, but I thought that you should also take a look.

cfriedt, 2023-07-07:
If it's a false positive, just remember to close it in coverity as well as on GH.

closed 2023-09-12 by jhedberg