zephyrproject-rtos/zephyr #59564: [Coverity CID: 321107] Copy of overlapping memory in subsys/shell/backends/shell_mqtt.c

zephyrbot, 2023-06-21:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/ce3317d03e46e1a218af3e712383d9216c48a992/subsys/shell/backends/shell_mqtt.c

Category: Memory - corruptions
Function: `init`
Component: Shell
CID: [321107](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=321107)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/ce3317d03e46e1a218af3e712383d9216c48a992/subsys/shell/backends/shell_mqtt.c#L658

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERLAPPING_COPY). ()

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

fabiobaltieri, 2023-08-22:
@jakub-uC @rlubos could you look into this?

rlubos, 2023-08-25:
I've checked the report, but it's a false-positive. There's no overlapping here, `sh_mqtt->holds` a NULL terminated string of known size (guaranteed above), and `sh_mqtt->sub_topic` is large enough to store that string + extra characters, as specified in the format. Closing.

closed 2023-08-25 by rlubos

fabiobaltieri, 2023-08-25:
@nashif can you flag this one?

rlubos, 2023-08-25:
@fabiobaltieri Do you mean like in Coverity? I marked it already as false-positive.

fabiobaltieri, 2023-08-25:
oh, thanks @rlubos, I was assuming you had no access to the tool for some reason, nevermind then