zephyrproject-rtos/zephyr #59594: npcx9m6f_evb: kernel/timer/timer_behavior/ failing

pkunieck, 2023-06-22:
**Describe the bug**
Tests from kernel/timer/timer_behavior/ failing, not a single test case pass. 

**To Reproduce**
Steps to reproduce the behavior:
  1. cd <path/to>zephyrproject/zephyr
  2. source zephyr-env.sh
  3. twister --device-testing --hardware-map <path/to>/map.yaml -T tests/kernel/timer/timer_behavior/ -vv

**Expected behavior**
The tests pass

**Impact**
annoyance

**Logs and console output**
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.2.0-2474-g1aebcec02ff9 ***
DEBUG   - DEVICE: Running TESTSUITE timer_jitter_drift
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_jitter_drift
DEBUG   - DEVICE: periodic timer behavior test
DEBUG   - -- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/ztest/zephyrproject/zephyr/twister-out/npcx9m6f_evb/zephyr/tests/kernel/ti

DEBUG   - DEVICE: periodic timer samples gathered, calculating statistics
DEBUG   - DEVICE: timer clock rate 15000000, kernel tick rate 1000, expected period 1000 us, samples 7000, period 1
DEBUG   - DEVICE: statistics samples 7000, rollovers 0, mean 2198.162029 us, variance 138.655087 us, stddev 11.7751
DEBUG   - DEVICE: statistics samples 7000, rollovers 0, mean 32972.430429 cycles, variance 31197.394588 cycles, std
DEBUG   - DEVICE: timer start cycle 310370, end cycle 231117383, total time 15387134.200000, expected time 7000000.
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:195: 
DEBUG   - DEVICE: Longest timer period too long (off by more than expected 10%)
DEBUG   - DEVICE: FAIL - test_jitter_drift in 16.093 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: TESTSUITE timer_jitter_drift failed.
DEBUG   - DEVICE: Running TESTSUITE timer_tick_train
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_one_tick_timer_train
DEBUG   - DEVICE: Initializing 4 Timers, Tick Rate 1000Hz, Expecting 2500 callbacks in 11000 ms
DEBUG   - DEVICE: Starting Timers with Skews
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 0
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 55
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 110
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 165
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 220
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 274
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 329
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 384
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 439
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 493
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 548
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 603
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 658
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 712
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 767
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 822
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 877
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 931
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 986
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1041
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1096
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1150
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1205
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1260
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1315
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1369
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1424
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1479
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1534
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1588
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1643
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1698
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1753
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1807
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1862
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1917
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 1972
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 2027
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 2081
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 2136
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 2191
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 2246
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 2300
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 2355
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 2410
DEBUG   - DEVICE: Faking busy work, remaining timers is 4, timer callbacks 2465
DEBUG   - DEVICE: One Tick Timer Train Done, took 11683 ms, busy loop ran 46 times
DEBUG   - DEVICE: Timer 0 max delta 69654 cycles or 4644 us
DEBUG   - DEVICE: Timer 1 max delta 69681 cycles or 4645 us
DEBUG   - DEVICE: Timer 2 max delta 69662 cycles or 4644 us
DEBUG   - DEVICE: Timer 3 max delta 69657 cycles or 4644 us
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/tick_timer_train.c:1
DEBUG   - DEVICE: Expected timer train to finish in under 11000 milliseconds, took 11683
DEBUG   - DEVICE: FAIL - test_one_tick_timer_train in 11.734 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: TESTSUITE timer_tick_train failed.
DEBUG   - DEVICE: 
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY START ------
DEBUG   - DEVICE: 
DEBUG   - DEVICE: SUITE FAIL -   0.00% [timer_jitter_drift]: pass = 0, fail = 1, skip = 0, total = 1 duration = 16.
DEBUG   - DEVICE: - FAIL - [timer_jitter_drift.test_jitter_drift] duration = 16.093 seconds
DEBUG   - DEVICE: 
DEBUG   - DEVICE: SUITE FAIL -   0.00% [timer_tick_train]: pass = 0, fail = 1, skip = 0, total = 1 duration = 11.73
DEBUG   - DEVICE: - FAIL - [timer_tick_train.test_one_tick_timer_train] duration = 11.734 seconds
DEBUG   - DEVICE: 
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY END ------
DEBUG   - DEVICE: 
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: RunID: 9fff733e18c4bd6078438aad2254fdbf
DEBUG   - DEVICE: PROJECT EXECUTION FAILED
DEBUG   - run status: npcx9m6f_evb/zephyr/tests/kernel/timer/timer_behavior/kernel.timer.timer failed

**Environment:**
 - OS: Linux
 - Toolchain Zephyr SDK 16.0
 - Commit: tested on the newest and older, always the same issue

FRASTM, 2023-06-23:
also found  failures on stm32 boards on this testcase :  https://github.com/zephyrproject-rtos/zephyr/issues/57285 and https://github.com/zephyrproject-rtos/zephyr/issues/57288

teburd, 2023-09-19:
> **Describe the bug** Tests from kernel/timer/timer_behavior/ failing, not a single test case pass.
>
> **To Reproduce** Steps to reproduce the behavior:
>
>     1. cd <path/to>zephyrproject/zephyr
>
>     2. [...]

This looks like the timer clock rate being used to calculate things is incorrectly set, and the actual timer clock source is a 32768 Hz clock?

teburd, 2023-10-17:
@MulinChao @ChiHuaL

MulinChao, 2023-10-23:
@teburd Sorry for the late reply. Need more time to dig it further. Any update will inform you.

ChiHuaL, 2023-10-24:
The default CPU runs at 15 MHz on the npcx9m6f_evb. 
By increasing the CPU clock to 90 MHz, it can pass almost all of the test criteria but failed in the final test item:
```
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:294: do_test_us)
Drift (in microseconds) outside expected bound
```
In the NPCX timer driver, it uses two hardware timer modules for the Zephyr kernel:
1. ITIM64 timer (system timer): It is a 64-bit timer which sources from APB2  clock (15 MHz) to records the absolute timestamp. It will always count without stopping during the test.
2. ITIM32 timer (event timer): it is the 32-bit event timer and can be configured to generate the interrupt to process the timeout event.
Note that, the event timer is default set to source from the  LFCLK (32.768 KHz) clock to support the low power mode (for `CONFIG_PM`)
As the test doesn't enable PM and In order to better observe the failure test item, we change its clock source from LFCLK to APB2.   
here is the test log:
```
*** Booting Zephyr OS build v3.5.0-rc3-86-g9a5dc7a251c4 ***                                                                                                                                                                                   
Running TESTSUITE timer_jitter_drift                                                                                                                                                                                                          
===================================================================                                                                                                                                                                           
START - test_jitter_drift_timer_period                                                                                                                                                                                                        
periodic timer behavior test using built-in restart mechanism                                                                                                                                                                                 
collecting time samples for approx 7 seconds                                                                                                                                                                                                  
periodic timer samples gathered, calculating statistics                                                                                                                                                                                       
timer clock rate 15000000, kernel tick rate 1000                                                                                                                                                                                              
period duration statistics for 7000 samples (0 rollovers):                                                                                                                                                                                    
  expected: 1000 us,            15000.000000 cycles                                                                                                                                                                                           
  min:      1023.800000 us,     15357 cycles                                                                                                                                                                                                  
  max:      1023.800000 us,     15357 cycles                                                                                                                                                                                                  
  mean:     1023.800000 us,     15357.000000 cycles                                                                                                                                                                                           
  variance: 0.000000 us,        0.000000 cycles                                                                                                                                                                                               
  stddev:   0.000000 us,        0.000000 cycles                                                                                                                                                                                               
timer start cycle 432171, end cycle 107931171,                                                                                                                                                                                                
total time 7166600.000000 us, expected time 7000000.000000 us,                                                                                                                                                                                
expected time drift 0.000000 us, difference 166600.000000 us                                                                                                                                                                                  
[... cut ...]
 - PASS - [timer_tick_train.test_one_tick_timer_train] duration = 10.453 seconds

------ TESTSUITE SUMMARY END ------

```
As we can see in the log, it consistently takes 1023 us for every sample period.
Because the extra 23 us should come from the interrupt delay (When the interrupt occurred, the ITIM64 timer still counts), we are wondering if  a fixed value (1000 us) for the criterion `TIMER_TEST_MAX_DRIFT`  without considering the interrupt delay and the number of `TIMER_TEST_SAMPLES`  make sense or not?  Did we misunderstand anything?

teburd, 2023-10-24:
This looks shockingly like a bug this test was meant to root out from the intel_adsp_timer driver.

Interrupt delays should not effect the timers. If they do that signifies a bug in the timer driver. That's exactly the kind of bug this test is meant to find and root out (they are hard!)

Take a very close look at the intel_adsp_timer.c, particularly around announce(), elapsed(), cycle_get_32() and cycle_get_64() and compare how that works with the npcx_itim_timer

I'm certain you'll find a small difference in the operational theory

PR #64715 (2023-11-17): driver: timer: npcx: fix the failure of the timer_behavior testsuite
The `cyc_sys_announced` (tick to be announced) was updated with the current timer will cause the time drift and induce the jitter when running the timer_behavior test suite. This PR fixed it by updating it at the tick boundary). The sys_clock_set_timeout API also changed because the above modification.
Also, as the npcx event timer is running at 32768 Hz (to support the low power sleep mode), it might introduce the deviation of (+/-) 30.5 us when converting the system time to event count. This PR increases the stdev tolerance [... cut]
commit 317d0702 (2023-11-17): tests: timer_behavior: increase stdev tolerance for npcx timer

Change the stdev tolerance stdev from 10 to 33 when npcx timer is used.
This is because the clock source of npcx event timer, which is used to
generate the timeout interrupt, is running at 32.768 KHz.
(i.e. 1 count = ~30.5 us.) The conversion from the absolute system timer
to the event timer count might have -30.5 ~ 30.5 deviation.
The tolerance setting is under the assumption that test sampes are
evenly located at 1030.5 and 969.5 for the worst case.

Fixes #59594
commit 4cfd4b53 (2023-11-17): driver: timer: npcx: fix announce/set timer timeout tick

The timer driver doesn't annouce/set the timeout at the tick boundary
but at the absolute next expiration time.
It will cause the accumatlation of the tick drift and cannot pass the
kernel/timer/timer_behavior test suite.
This commit fixes the tick drift problem by annouce the time at the tick
bouandry.

Fixes #59594
closed 2023-11-17 by fabiobaltieri